realthunder/FreeCAD_assembly3 #724: [Question] Recompute assembly from macros

Shkolik, 2021-07-10:
Hello! 

I'm trying to write macro to animate assembly using PointDistanceConstraint and updating positions of elements under.
So fr I was able to change position of element, but struggle to recompute assembly and update view. 
I'm doing this:

`FreeCAD.ActiveDocument.getObject('Element005').Offset = App.Placement(App.Vector(self.x_pos,0,0),App.Rotation(App.Vector(0,0,1),0))
FreeCAD.ActiveDocument.recompute()`

Please point me in right direction how to force recompute and update view. If I right click on assembly and then 'Recompute object' everything gets updated as expected.

Thank you!

realthunder, 2021-07-11:
If you have asm3 auto recompute enable, the assembly will be solved on recompute. Otherwise, you can manually invoke the solve command as below

```
Gui.runCommand('asm3CmdQuickSolve')
```

When you press any toolbar button, pay attention to what's printed in the Python console. You'll find clues about how to invoke the command for that button.

Shkolik, 2021-07-11:
Thank you a lot! asm3CmdQuickSolve works like a charm! Just for your interest - auto recompute enabled but FreeCAD.ActiveDocument.recompute() does not force view to redraw and assembly to solve.

closed 2021-07-11 by Shkolik