realthunder/FreeCAD_assembly3 #725: Unable to disable "Draft tray" Toolbar

StudioPetrikas, 2021-07-10:
"Draft tray" Toolbar keeps appearing when whether Arch or Draft Workbenches get activated. 
Even if it's disabled in Parameters or via View -> Toolbars menu, it reappears when the Arch or Draft workbenches get activated.

realthunder, 2021-07-12:
Can you please send me your `link.user.cfg` file here? On Windows, it is in `c:\users\<username>\AppData\Roaming\FreeCAD`. On Linux, it is in `~\.FreeCAD`.

StudioPetrikas, 2021-07-12:
I can reproduce this on a clean setup (don't have my computer on hand)

To reproduce:
- Activate Arch or Draft workbench
- Disable "Draft tray" toolbar through View -> Toolbars
- Activate any other workbench
- Activate Arch or Draft workbench
- "Draft tray" should reappear. 

I think in Vanilla FreeCAD we don't even have the ability to disable it through View -> Toolbars. I think you added that functionality. Thank you for that!

StudioPetrikas, 2021-07-18:
This is closed! Thank you so much, I can now continue with my UI project :)

closed 2021-07-18 by StudioPetrikas