realthunder/FreeCAD_assembly3 #726: No cursor in python console when in overlay mode

Shkolik, 2021-07-11:
Hello again. I was tinkering with python scripting last few days and found minor issue in python console overlay. There is no cursor when clicking on console line to enter command, so it looks like console in read-only mode. In standard view everything ok. See recording.

https://user-images.githubusercontent.com/26121979/125207653-00e36e80-e253-11eb-8fda-07156d34b125.mp4

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

realthunder, 2021-07-12:
I am aware of this problem. Unfortunately, this is a not easy problem to solve. If you are really bothered by this problem, you can turn off the `Transparent` mode of the panel, so that it become opaque when mouse over, and revert to transparent when leave.

[image]

Shkolik, 2021-07-12:
Good to know that it's known issue and it has workaround. Not a deal breaker, at least for me. Thank you!

PS: Will leave issue opened just to keep track.