realthunder/FreeCAD_assembly3 #727: Crash removing all contraints from sketch: body with only two sketches nothing else

apper002, 2021-07-11:
OS: Ubuntu 20.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.615.24301 +3747 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.9.4
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7fb9cf83d210]
#1  /tmp/.mount_Free-0yEzKZD/usr/lib64/SketcherGui.so(+0x19acc3) [0x7fb9aa700cc3]
#2  0x7fb9aa71a988 in SketcherGui::ViewProviderSketch::updateColor() from /tmp/.mount_Free-0yEzKZD/usr/lib64/SketcherGui.so+0x1528
#3  0x7fb9aa71f859 in SketcherGui::ViewProviderSketch::onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_Free-0yEzKZD/usr/lib64/SketcherGui.so+0x269
#4  0x7fb9d1fbd9a2 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/libFreeCADGui.so+0xf2
#5  /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/libFreeCADGui.so(+0x74176f) [0x7fb9d1c8076f]
#6  0x7fb9d1fbdc13 in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/libFreeCADGui.so+0x253
#7  /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/libFreeCADGui.so(+0x74176f) [0x7fb9d1c8076f]
#8  0x7fb9d1fbe9ee in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/libFreeCADGui.so+0x1ee
#9  0x7fb9d1fcb0e8 in Gui::SelectionSingleton::setPreselect(char const*, char const*, char const*, float, float, float, int, bool) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/libFreeCADGui.so+0x658
#10  0x7fb9aa702c43 in SketcherGui::ViewProviderSketch::detectPreselection(SoPickedPoint const*, Gui::View3DInventorViewer const*, SbVec2s const&, bool) from /tmp/.mount_Free-0yEzKZD/usr/lib64/SketcherGui.so+0x393
#11  0x7fb9aa733ecf in SketcherGui::ViewProviderSketch::mouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /tmp/.mount_Free-0yEzKZD/usr/lib64/SketcherGui.so+0x6ef
#12  0x7fb9d1e733d7 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/libFreeCADGui.so+0x1f7
#13  0x7fb9cf3d76ef in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/../lib/libCoin.so.80+0xbf
#14  0x7fb9cf1d1149 in SoAction::traverse(SoNode*) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#15  0x7fb9cf392a3a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/../lib/libCoin.so.80+0x16a
#16  0x7fb9cf3e05ea in SoGroup::doAction(SoAction*) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/../lib/libCoin.so.80+0x5a
#17  0x7fb9cf3fc80b in SoSeparator::doAction(SoAction*) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/../lib/libCoin.so.80+0x2b
#18  0x7fb9cf1d1149 in SoAction::traverse(SoNode*) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#19  0x7fb9cf392a3a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_Free-0yEzKZD/usr/bin/../lib64/../lib/libCoin.so.80+0x16a

realthunder, 2021-07-12:
I can't reproduce the problem. How do you delete all constraints. Can you post a screencast showing you steps, and provide a sample file showing the problem. Thanks.

apper002, 2021-07-12:
Hi Zheng,

 

It's a minor issue. Attached the Sketch.

 

Just delete all contraints from 1st Sketch and then recompute.

 

Rgds

Aart

 
 

Gesendet: Montag, 12. Juli 2021 um 03:14 Uhr

luzpaz, 2021-07-12:
We never got the attachment. Please use the gituhub UI to respond

luzpaz, 2022-07-01:
@apper002 please follow up