realthunder/FreeCAD_assembly3 #728: Feature request: cutting lines in Clipping plane

pgilfernandez, 2021-07-12:
Hi,

The new "solid mode" in Clipping plane is great, thanks!
Would it be possible to add an option to draw the lines  in that solid plane? sometimes it is still hard to differentiate what is the part and what is the cutted plane...

Example:
[image]

Thanks

realthunder, 2021-07-12:
Yes, it is possible, but it is currently not possible to do it only for the clipping plane. This will enable outline drawing on all faces. That's why it is only enabled in `Hidden line` draw style. Besides, the outline drawing is not very efficient at the moment. You may experience some slow down. You can simply enable `Invert fill color` to make the clip plane stand out.

[image]

herrgerd, 2021-07-12:
> 
> 
> Hi,
> 
> The new "solid mode" in Clipping plane is great, thanks! (...)
> Thanks

**Indeed it is!** Had to use Inventor a lot lately and clipping parts and especially assemblies is a nightmare there. Still using some strange blurred "pattern" which makes recognizing details impossible at times. You have to create your own pattern and import it as .bmp to change that. No solid color sectional view possible via on-board means, in a multi thousand $ software suite, go figure :)

mwlandolsi, 2021-07-16:
Another enhancement, is  to define for every part or body it's hashing style, it could be easier for every surface. This is very important in mechanical engineering.

OxygenLiu, 2021-07-28:
A noob question, how to activate the Solid Clipping Plane?

mwlandolsi, 2021-07-28:
@OxygenLiu  go to **View> Clipping plane** menu.

OxygenLiu, 2021-07-30:
@mwlandolsi no option for solid clipping plane as shown below 

[image]

FreeCAD version:

OS: Pop!_OS 20.04 LTS (pop:GNOME/pop)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-07-30:
It is only available in the `Daily` branch.