realthunder/FreeCAD_assembly3 #729: [Features Requests] Point on B-spline edge and spline trimming

YosypC, 2021-07-13:
[image]

I guess the picture is self explanatory.
I also need to pad only a section of the exported geometry, but currently I can't trim the B-spline cause I have no point to do so.

oficinerobotica, 2021-07-18:
The sketcher development is mainly done by Abdulah and this issue might be on the roadmap for solving with the latest sketcher developments https://forum.freecadweb.org/viewtopic.php?f=10&t=51716. It is also a feature that I often need and wish we would have. 

Please make a feature request here: https://forum.freecadweb.org/viewforum.php?f=8 with the title "[Feature request] bspline tangency in sketcher" or other suitable title and I will join the conversation there trying to ping the right people. 

Then we can close this issue here.

YosypC, 2021-07-19:
Post submitted, waiting moderator approval.

closed 2021-07-19 by YosypC