realthunder/FreeCAD_assembly3 #731: [Question] Line width on different resolutions

Shkolik, 2021-07-13:
Hello!

Found interesting thing that I don't know is it feature or bug or something else, or maybe there is workaround.

I have 2 laptops - one with 4K resolution, another with 1080p. I'm working on both and sync my work using github. The setting are the same across laptops, except Line width in PartDesign settings and maxWidth in Display setting. It's obvious that on smaller resolution I need thinner lines. 

So now when I editing file that was created on 4K monitor on HD display it looks awful. 

4K:
[image]
[image]

FHD:
[image]
[image]

Any warkaround available to respect maxWidth of the line and vertex? Maybe it has to be some checkbox for it?

PS: Of course new bodies will be created with respect to this settings.

realthunder, 2021-07-14:
I added a new option in `Preferences -> PartDesign -> Shape appearence -> Respect system DPI`. Enable this to get auto scale for line width and point size.

Shkolik, 2021-07-14:
Wow! That's fast! Will test when it will be integrated in next daily release. Thank you!

Shkolik, 2021-07-18:
Hm, just tested on 0717 daily. Looks better, but still not ideal. Looks like this checkbox only reduce line width to the minimum - not to default line size that set in settings. So as temporary solution it's ok, but little bit confusing, because all new solids will have another line width. 

[image]