realthunder/FreeCAD_assembly3 #732: Build next daily with OCCT 7.5.1+

DevyHeavy, 2021-07-15:
In the current build of FreeCAD, the step exporter generates incorrect geometry.

When doing subsequent builds, maybe the OCCT version can be increased to 7.5.2 or above (7.5.1 is also broken).

See: https://github.com/CadQuery/cadquery/issues/697
And: https://forum.freecadweb.org/viewtopic.php?f=22&t=57735

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)
```

luzpaz, 2021-07-30:
@realthunder is LS3 being built with OCC v7.5.2 ATM? (see https://github.com/CadQuery/cadquery/issues/697#issuecomment-810171641)

realthunder, 2021-07-31:
Windows release uses LibPack which is still at 7.5.0. Are there any update?

luzpaz, 2021-07-31:
CC @sgrogan are there plans to update the Windows LibPack to 7.5.2 ?

sgrogan, 2021-08-02:
Recently @apeltauer has been maintaining the LibPack, but I will look when I have a chance. Usually a OCCT update requires a simultaneous netgen update.

apeltauer, 2021-08-02:
I can see that 7.5.0 is the latest stable release. Is this correct?

sgrogan, 2021-08-02:
I guess they call it a "maintenance release" I guess like FreeCAD uses "backports"?
https://dev.opencascade.org/content/open-cascade-technology-752-maintenance-release

apeltauer, 2021-08-03:
ah, ok. The latest libpack has also a fillet fix from https://gitlab.com/blobfish/occt provided by  [Chris Hennes](https://gitlab.com/chennes)
Is this already in the version 7.5.2?

luzpaz, 2021-08-03:
@apeltauer yup

apeltauer, 2021-08-03:
https://github.com/apeltauer/FreeCAD/releases/tag/LibPack_12.5.3
--> new libpack with occ 7.5.2
--> i have used the all binaries from the libpack 12.5.2 (including netgen)

chennes, 2021-08-03:
> @chennes ah, ok. The latest libpack has also a fillet fix from https://gitlab.com/blobfish/occt provided by [Chris Hennes](https://gitlab.com/chennes)
> Is this already in the version 7.5.2?

The official OCCT 7.5.2 does not yet incorporate the patch I submitted (as far as I am aware) -- I believe it is scheduled for the 7.6 release.

luzpaz, 2021-08-03:
@chennes [that's right](https://tracker.dev.opencascade.org/view.php?id=32181). That slipped my mind.

apeltauer, 2021-08-07:
see FC thread: https://forum.freecadweb.org/viewtopic.php?p=522475#p522475
@chennes occ 7.5.3 also does not include your fix right? can you/we make a hotfix on your fork?

luzpaz, 2021-08-07:
@apeltauer why not use @blobfish's hotfix?

chennes, 2021-08-07:
@apeltauer Maybe you can just apply the diff from OCCT's repo? https://git.dev.opencascade.org/gitweb/?p=occt.git;a=commitdiff;h=329e5df986519164083d78b28ae62d1a3b10fa6a;hp=7fd4958d451a9c8dcf1c58e100c8ac898484addd

I'd expect it to apply cleanly, the changes are very small and only relate to error handling.

blobfish, 2021-08-07:
> @apeltauer Maybe you can just apply the diff from OCCT's repo? [...]

FYI: I reverted chennes changes on my branch, as he was working with occt directly to get his fixes in. So I just figured I would get his final changes after an occt release and I wouldn't have any rebase conflicts.

luzpaz, 2022-01-12:
Latest LinkDaily: (shows OCC v7.5.3)
```
OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

luzpaz, 2022-04-08:
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

closed 2022-04-08 by luzpaz