realthunder/FreeCAD_assembly3 #735: PartDesign boolean split Can not choose which side to keep

parnasist3, 2021-07-16:
When performing a split of a solid via Boolean operation, you do not seem to get any option to "flip side"?

[image]

[PartDesign_Arbitrary_surface.FCStd.txt](https://github.com/realthunder/FreeCAD_assembly3/files/6831292/PartDesign_Arbitrary_surface.FCStd.txt)

oficinerobotica, 2021-07-18:
I think a better tool for what you want to achieve would be the split tool right next to the PartDesign boolean tools. It was especially designed to work with surfaces that split solids. It allows activation/deactivation of the splits by double clicking on the split components in the tree. 

The only caveat is that your surface needs to fully intersect the solid. For some reason, with your file, the surface creates only one split. Might be because of coplanar/colinear geometry of the boolean components. You could create a copy of your surface that extends in order to fully cut your intended solid. I made an example video with a datum plane as cutting surface. 

Does this solve your issue?

https://user-images.githubusercontent.com/26909793/126054014-90174240-a746-4c0c-96ac-539b5075fffa.mp4

nguyenhaiac, 2021-07-20:
I have the same issue with my design that the split tool can only generate one part when using split file
[sofle-rgb.FCStd.txt](https://github.com/realthunder/FreeCAD_assembly3/files/6849231/sofle-rgb.FCStd.txt)
. Can you please help? @oficinerobotica

realthunder, 2021-07-20:
I didn't find any split usage in your file, @nguyenhaiac?

nguyenhaiac, 2021-07-21:
> I didn't find any split usage in your file, @nguyenhaiac?

Oh, I think I deleted the split feature when it didn't work. I tried to use to split the Body body with the latest feature of the Separation body (thickness). I've change the thickness features and it works now, the result is 2 split part (the thickness enclosed a part of the body before, and that part is cut out when using split. )

Before
[image]

After

[image]

parnasist3, 2021-08-08:
@oficinerobotica:
> ... the split tool right next to the PartDesign boolean tools. It was especially designed to work with surfaces to split solids. It allows activation/deactivation of the splits by double clicking on the split components in the tree.

> Does this solve your issue?
>
Sorry for late reply (summer heat happened).
Tried your suggestion and it works if, and only if, the _plane_ is created inside the same Body where you will perform your splitting operation. With PartDesign plane creation tool, that is.
However, I can't make it work with an offset (which doesn't touch the solids edges) of my double curved surface.

Body002 contains the object to be split, but the surface and plane will never ever be situated in a Body of any kind. Ever.
So, here we are lacking some automation for making an automated and linked "thingie" to use as splitting surface with Body002.

It does not work with this version either:

OS: openSUSE Leap 15.2 (KDE//usr/share/xsessions/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2

parnasist3, 2021-08-29:
@realthunder: any plans to make this "_use a plane or surface from anywhere to cut anything_" come true?

realthunder, 2021-08-30:
@parnasist3 The `...anywhere...anything` is probably a bit too broad of a scope. My current aim is to gradually remove restriction of using external features inside a body. The above mentioned commit shall allow you to use external faces freely in a PartDesign split.