realthunder/FreeCAD_assembly3 #736: Draft not working

MarcelP102, 2021-07-16:
Can't draw with any of the draft tools. Entering coordinates manual in Task panel does work..

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.716.24301 +3839 (Git)
Build type: Release
Branch: LinkDaily
Hash: 31965a2cdaa4a5870b9608ad58c25115be334f13
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Dutch/Netherlands (nl_NL)

herrgerd, 2021-07-16:
Ummmmm, same for me.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.716.24301 +3839 (Git)
Build type: Release
Branch: LinkDaily
Hash: 31965a2cdaa4a5870b9608ad58c25115be334f13
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

mwlandolsi, 2021-07-16:
Hello,
I do not know if this fix can also solve the other problem in draft which is object manipulation (translate, rotate ...) with reference to a vertex by snapping to the nearest for example.

realthunder, 2021-07-16:
Can you post a sample file showing the problem?

mwlandolsi, 2021-07-17:
sorry to be little bit late, I was busy the hole day. the issue seems to be solved with today version 07172021 I was talking about this functionality in Draft WB. 

https://user-images.githubusercontent.com/76944804/126043430-543d42dd-c601-439d-9720-c10c7f97de50.mp4

oficinerobotica, 2021-07-18:
@MarcelP102 With the 07.17release would you consider this issue solved? If so please close the issue.

mwlandolsi, 2021-07-18:
after testing some other functionalities, looks there are some other bugs.

https://user-images.githubusercontent.com/76944804/126073289-73e21cc1-e667-456b-93c4-041dda5fead4.mp4

[draft snap test.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6836527/draft.snap.test.FCStd.zip)

MarcelP102, 2021-07-18:
@mwlandolsi  perhaps you can open a new issue?

closed 2021-07-18 by MarcelP102