realthunder/FreeCAD_assembly3 #738: [Experimental renderer] Selection on top doesn't work on hidden lines and is always activated on asm3 mover tools.

oficinerobotica, 2021-07-18:
The selection on top is working correctly if the renderer is set to auto. With experimental, the hidden lines/faces can't be selected.

https://user-images.githubusercontent.com/26909793/126055329-817ecc63-f4f0-4022-abf8-9c7ff3f12345.mp4

Instead the mover tools issue is present with both auto and experimental. Independent of how "selection on top" is activated, now all the hierarchy will be rendered as on top. In the previous daily the issue was present but the transparent rendering stopped at the boundary of the  assembly containing the selection.

https://user-images.githubusercontent.com/26909793/126055401-4c17174f-5280-4584-bcb7-1599320548c8.mp4

realthunder, 2021-07-18:
The hidden line selection on top is control by an option in `Preference -> General -> Selection -> Enable hidden line selection`. Maybe you disable it somehow?

The mover problem can be solved by disabling the `Show on top` option in the task panel. This disables the whole assembly on top, but still keeps the moving part on top. If you don't want that on top either, simply press `Ctrl + Shift + Space` when there is no selection to clear all on top objects.

[image]

realthunder, 2021-07-18:
BTW, regardless of option `Hidden line selection in on top object`, you can always press `Shift` to enable hidden element selection, including `Face`. You can also use `Shift + Ctrl + Wheel` to cycle through hidden elements.

oficinerobotica, 2021-07-18:
@realthunder the `Preference -> General -> Selection -> Enable hidden line selection` is enabled and working correctly when renderer is set to auto. It misbehaves when set to experimental blocking hidden line selection. 

Unfortunately for me Shift to enable hidden element is clashing with the touch pad navigation mode for viewport pan. I realize that accounting for all navigation modes shortcuts is impossible and the shortcut that you set should stay unchanged. I'm perfectly happy with the way selection on top works when in auto renderer mode. Also there is G, G to help.

About the asm3 transparency. My bad, that show on top just skipped me and I can't remember when I enabled it. Your solution works perfectly. Sorry for the noise.

realthunder, 2021-07-19:
Just pushed a fix for this. Apparently the problem only happens if you disable `Pre-selection on top` option. Just curious, any particular reason you disabled `Pre-selection on top` and `Draw pre-selection outline`?

oficinerobotica, 2021-09-22:
Sorry, somehow I forgot to close this.  To answer your previous question, I'm constantly changing between selection/preselection modes during modeling. I have pie menus setted up for fast changing of modes because it is such a godsend to have the flexibility. I change based on the visual clutter of the scene or if I have to select back faces/edges. During a modeling session it might change a million times

closed 2021-09-22 by oficinerobotica