realthunder/FreeCAD_assembly3 #739: Perspective view sketcher labels rendering issue.

oficinerobotica, 2021-07-18:
This is less than a minor issue but I thought I should mention it given the development of the experimental renderer.
If the view is set to perspective, when in sketcher, the labels will be rendered out of proportion. This happens with both auto and experimental renderer option. I also tested in Master and there the issue is exactly the opposite.  The labels will be rendered really small almost disappearing
[image]
.

realthunder, 2021-07-18:
I am aware of this problem, and has [this](https://github.com/realthunder/FreeCAD/commit/8e2e2d34cfbe97a75d05100f83ae9890d6e9911f) commit to specifically fix this problem. It looks okay on my display. But apparently it is not fully solved. Can you post your display details here, like resolutions, any DPI scaling involved?

herrgerd, 2021-07-18:
I just tried it (never using perspective, so I had to first find where to activate it :) ) and it seems to change size according to view angle/position:
[image]

Display Settings:
[image]

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.716.24301 +3839 (Git)
Build type: Release
Branch: LinkDaily
Hash: 31965a2cdaa4a5870b9608ad58c25115be334f13
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

oficinerobotica, 2021-07-18:
I don't know if it makes any difference but I just realized that the Xserver info reports a wrong monitor size in mm in the info panel. 
Real size 1920x1080px 310x175mm 14 inch monitor.  Scale 100%
[image]
OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)