realthunder/FreeCAD_assembly3 #741: [Bug] Crash when closing a file

Samantha-uk, 2021-07-19:
**Issue**
App crashes when closing a file. 
[image]

**Steps to reproduce**
1. Start FreeCad
2. File->New
3. File->Close

 
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

thomasgi1, 2021-07-19:
Crashes here too:

Process:               FreeCAD [6014]
Path:                  /Applications/FreeCAD asm3.app/Contents/Resources/bin/FreeCAD
Identifier:            ???
Version:               ??? (2021)
Code Type:             X86-64 (Native)
Parent Process:        ??? [6007]
User ID:               1002

Date/Time:             2021-07-19 18:36:30.004 +0200
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Anonymous UUID:        5D9CF24A-AD57-94ED-B1A7-307DCFC9C006

Sleep/Wake UUID:       2F9269A6-5EFF-4D42-806A-D61EA94343B6

Time Awake Since Boot: 780000 seconds
Time Since Wake:       540000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [6014]

VM Regions Near 0x10:
--> 
    __TEXT                      10216b000-102172000    [   28K] r-x/r-x SM=COW  /Applications/FreeCAD asm3.app/Contents/Resources/bin/FreeCAD

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libCoin.80.0.1.dylib          	0x0000000104d93e10 SoCallbackAction::~SoCallbackAction() + 80
1   libCoin.80.0.1.dylib          	0x0000000104d93fbe SoCallbackAction::~SoCallbackAction() + 14
2   libFreeCADGui.dylib           	0x000000010252b24d SoFCRenderCacheManagerP::~SoFCRenderCacheManagerP() + 29
3   libFreeCADGui.dylib           	0x000000010252b501 SoFCRenderCacheManager::~SoFCRenderCacheManager() + 33
4   libFreeCADGui.dylib           	0x000000010255ad48 Gui::SoFCUnifiedSelection::~SoFCUnifiedSelection() + 40
5   libFreeCADGui.dylib           	0x000000010255adce Gui::SoFCUnifiedSelection::~SoFCUnifiedSelection() + 14
6   libCoin.80.0.1.dylib          	0x0000000104f7080c SoBase::destroy() + 172
7   libCoin.80.0.1.dylib          	0x0000000104f63f1a SoBaseList::truncate(int) + 74
8   libCoin.80.0.1.dylib          	0x0000000104f74d01 SoChildList::~SoChildList() + 17
9   libCoin.80.0.1.dylib          	0x0000000104fcbec5 SoGroup::~SoGroup() + 37
10  libCoin.80.0.1.dylib          	0x0000000104fe9b4e SoSeparator::~SoSeparator() + 14
11  libCoin.80.0.1.dylib          	0x0000000104f7080c SoBase::destroy() + 172
12  libCoin.80.0.1.dylib          	0x0000000104d9ba26 SoHandleEventAction::~SoHandleEventAction() + 54
13  libCoin.80.0.1.dylib          	0x0000000104d9ba9e SoHandleEventAction::~SoHandleEventAction() + 14
14  libCoin.80.0.1.dylib          	0x0000000104f90eb3 SoEventManager::~SoEventManager() + 243
15  libCoin.80.0.1.dylib          	0x0000000104f9111e SoEventManager::~SoEventManager() + 14
16  libFreeCADGui.dylib           	0x00000001026067bc SIM::Coin3D::Quarter::QuarterWidget::~QuarterWidget() + 252
17  libFreeCADGui.dylib           	0x000000010263f326 Gui::View3DInventorViewer::~View3DInventorViewer() + 1254
18  libFreeCADGui.dylib           	0x000000010263fd4e Gui::View3DInventorViewer::~View3DInventorViewer() + 14
19  libFreeCADGui.dylib           	0x0000000102633f64 Gui::View3DInventor::~View3DInventor() + 404
20  libFreeCADGui.dylib           	0x000000010263403e Gui::View3DInventor::~View3DInventor() + 14
21  libQt5Core.5.12.9.dylib       	0x000000010498087d QObject::event(QEvent*) + 719

OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

realthunder, 2021-07-21:
Can't reproduce on my Mac. Are you guys using the BigSur release? If so can you please check if the same thing happen with my Conda release?

thomasgi1, 2021-07-21:
I tried it under MacOS 10.14 (Mojave) with the same crash:

Process:               FreeCAD [15466]
Path:                  /Applications/FreeCAD asm3.app/Contents/Resources/bin/FreeCAD
Identifier:            ???
Version:               ??? (2021)
Code Type:             X86-64 (Native)
Parent Process:        ??? [15458]
Responsible:           FreeCAD [15466]
User ID:               501

Date/Time:             2021-07-21 15:35:48.137 +0200
OS Version:            Mac OS X 10.14.6 (18G9216)
Report Version:        12
Anonymous UUID:        00D0AAE4-4FA2-1AA2-1DC3-429829F4143B

Sleep/Wake UUID:       1D4E49A1-FE07-43C4-8AF5-FD90963B831D

Time Awake Since Boot: 880000 seconds
Time Since Wake:       170 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [15466]

VM Regions Near 0x10:
--> 
    __TEXT                 000000010e581000-000000010e588000 [   28K] r-x/r-x SM=COW  /Applications/FreeCAD asm3.app/Contents/Resources/bin/FreeCAD

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libCoin.80.dylib              	0x000000011119be10 SoCallbackAction::~SoCallbackAction() + 80
1   libCoin.80.dylib              	0x000000011119bfbe SoCallbackAction::~SoCallbackAction() + 14
2   libFreeCADGui.dylib           	0x000000010e93c24d SoFCRenderCacheManagerP::~SoFCRenderCacheManagerP() + 29
3   libFreeCADGui.dylib           	0x000000010e93c501 SoFCRenderCacheManager::~SoFCRenderCacheManager() + 33
4   libFreeCADGui.dylib           	0x000000010e96bd48 Gui::SoFCUnifiedSelection::~SoFCUnifiedSelection() + 40
5   libFreeCADGui.dylib           	0x000000010e96bdce Gui::SoFCUnifiedSelection::~SoFCUnifiedSelection() + 14
6   libCoin.80.dylib              	0x000000011137880c SoBase::destroy() + 172
7   libCoin.80.dylib              	0x000000011136bf1a SoBaseList::truncate(int) + 74
8   libCoin.80.dylib              	0x000000011137cd01 SoChildList::~SoChildList() + 17
9   libCoin.80.dylib              	0x00000001113d3ec5 SoGroup::~SoGroup() + 37
10  libCoin.80.dylib              	0x00000001113f1b4e SoSeparator::~SoSeparator() + 14
11  libCoin.80.dylib              	0x000000011137880c SoBase::destroy() + 172
12  libCoin.80.dylib              	0x00000001111a3a26 SoHandleEventAction::~SoHandleEventAction() + 54
13  libCoin.80.dylib              	0x00000001111a3a9e SoHandleEventAction::~SoHandleEventAction() + 14
14  libCoin.80.dylib              	0x0000000111398eb3 SoEventManager::~SoEventManager() + 243
15  libCoin.80.dylib              	0x000000011139911e SoEventManager::~SoEventManager() + 14
16  libFreeCADGui.dylib           	0x000000010ea177bc SIM::Coin3D::Quarter::QuarterWidget::~QuarterWidget() + 252
17  libFreeCADGui.dylib           	0x000000010ea50326 Gui::View3DInventorViewer::~View3DInventorViewer() + 1254
18  libFreeCADGui.dylib           	0x000000010ea50d4e Gui::View3DInventorViewer::~View3DInventorViewer() + 14
19  libFreeCADGui.dylib           	0x000000010ea44f64 Gui::View3DInventor::~View3DInventor() + 404
20  libFreeCADGui.dylib           	0x000000010ea4503e Gui::View3DInventor::~View3DInventor() + 14
21  libQt5Core.5.12.9.dylib       	0x0000000110d8b87d QObject::event(QEvent*) + 719

OS: macOS Mojave (10.14)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

Note that in contrast to MacOS BigSur the crash report isn't shown in Mojave. FC just silently dies. One has to open the Konsole app and look within the user reports in order to find the related crash dump. When starting FC from the command line you see at least the segfault:

/Applications/FreeCAD asm3.app/Contents/MacOS/FreeCADLink: line 27: 15587 Segmentation fault: 11  "$PREFIX/bin/$name" "$@"

On both machines  I used your [Conda release](https://github.com/realthunder/FreeCAD_assembly3/releases/download/0.11/FreeCAD-asm3-Daily-OSX-Conda-Py3-Qt5-20210717-x86_64.dmg) for testing.

luzpaz, 2021-07-21:
It is possible to use a debugger on macOS, would you be game, @thomasgi1 ?  
https://wiki.freecadweb.org/Debugging#For_MacOSX

thomasgi1, 2021-07-21:
I started FC in lldb, but due to the lack of debug information I cannot get into source level debugging. On weekend I tried building FC asm3 on my machine, but unfortunately it does not start due to a missing symbol. Thus, my own build is faulty.

realthunder, 2021-07-21:
@thomasgi1, have you tried with the BigSur release image?

thomasgi1, 2021-07-22:
It does NOT crash in BigSur release.

OS: macOS 11.4
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git)
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.9.4
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Germany (en_DE)

luzpaz, 2022-07-11:
@realthunder should we close ?

realthunder, 2022-07-13:
This problem is likely fixed [here](https://github.com/realthunder/coin/commit/41dd4382f6068dd824dd2f515b087bebeeeb066c).

closed 2022-07-13 by realthunder