realthunder/FreeCAD_assembly3 #743: [Path] 'PySide2.QtWidgets.QWidget' object has no attribute 'enableExtensionsWarning'

Sebclem, 2021-07-19:
Hi ! 

I don't know if it's linked to the assembly branch or if it's from the upstream:

When i try to use the `Adaptive Clearing` or `Path Pocket` from the Path workbench I get this error

```python
18:48:11  Traceback (most recent call last):
  File "C:\Users\seb65\Documents\Freecad\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210717\Mod\Path\PathScripts\PathOpGui.py", line 104, in setEdit
    self.setupTaskPanel(TaskPanel(vobj.Object, self.deleteObjectsOnReject(), page, selection))
  File "C:\Users\seb65\Documents\Freecad\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210717\Mod\Path\PathScripts\PathOpGui.py", line 1028, in __init__
    self.featurePages.append(opPage.taskPanelBaseLocationPage(obj, features))
  File "C:\Users\seb65\Documents\Freecad\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210717\Mod\Path\PathScripts\PathAdaptiveGui.py", line 247, in taskPanelBaseLocationPage
    self.extensionsPanel = PathFeatureExtensionsGui.TaskPanelExtensionPage(obj, features) # pylint: disable=attribute-defined-outside-init
  File "C:\Users\seb65\Documents\Freecad\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210717\Mod\Path\PathScripts\PathOpGui.py", line 201, in __init__
    self.form = self.getForm()  # pylint: disable=assignment-from-no-return
  File "C:\Users\seb65\Documents\Freecad\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210717\Mod\Path\PathScripts\PathFeatureExtensionsGui.py", line 230, in getForm
    form.enableExtensionsWarning.hide()
<class 'AttributeError'>: 'PySide2.QtWidgets.QWidget' object has no attribute 'enableExtensionsWarning'
```

Tested on Window and Linux (KDE Neon)

Just tell me if you think I need to open an issue on the upstream repo.

luzpaz, 2021-07-20:
Please always share the full [About](https://wiki.freecadweb.org/About) info of any FreeCAD build you've tested bugs on.

luzpaz, 2021-08-04:
@realthunder should I label this `waiting next release` ?

kkunzelm, 2021-11-14:
Hello,

I observe exactly the same problem as described by Sebclem. I use the lastest stable version of you Freecad Appimage release (see "About" infos at the end of this post). 

I would have assumed that the July 21 commits are already integrated in the October 15 version of your Appimage. 

The weekly build of the official Freecad Appimage (Version: 0.20.26306 (Git) AppImage) has no problems with Path Pockets or Adaptive Clearing (but they have a problem with Draft -> Array). But I do not want to mix use different versions of Freecad for my project. 

Would it help to compile your version of Freecad ( realthunder/FreeCAD ) myself to overcome the Path Workbench problem? 

Thanks a lot for your great works.

Sincerely

Karl-Heinz

OS: Debian GNU/Linux 11 (bullseye) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +3962 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: c6cb39697eb439caf7e5bd0208ca3082a7820547
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: German/Germany (de_DE)

realthunder, 2021-11-14:
The next release will sync my branch with recent upstream 0.20, which should solve this problem here.

luzpaz, 2022-01-12:
@Sebclem please retest on latest release

mnswamp1, 2022-02-07:
Hi, looks like this is still an issue with the 20220111 stable release.  I've included both the drawing and traceback here:  https://github.com/mnswamp1/linkstage3-issues
Windows 10
Latest stable version 20220111
Steps

I've tried this with pocket, face and adaptive.  No joy in an otherwise insanely functional update.

Thanks

marsh

luzpaz, 2022-07-11:
@mnswamp1 can you still recreate this on latest version?

Huwyler, 2022-08-28:
I hope you don't mind if I answer this: Not exactly the same use-case, but also in path::pocket, also 
"class 'AttributeError'>: 'PySide2.QtWidgets.QWidget' object has no attribute 'enableExtensionsWarning'

Full trace:
  File "/tmp/.mount_FreeCAYLstxT/usr/Mod/Path/PathScripts/PathOpGui.py", line 104, in setEdit
    self.setupTaskPanel(TaskPanel(vobj.Object, self.deleteObjectsOnReject(), page, selection))
  File "/tmp/.mount_FreeCAYLstxT/usr/Mod/Path/PathScripts/PathOpGui.py", line 1028, in __init__
    self.featurePages.append(opPage.taskPanelBaseLocationPage(obj, features))
  File "/tmp/.mount_FreeCAYLstxT/usr/Mod/Path/PathScripts/PathPocketShapeGui.py", line 62, in taskPanelBaseLocationPage
    self.extensionsPanel = PathFeatureExtensionsGui.TaskPanelExtensionPage(obj, features) # pylint: disable=attribute-defined-outside-init
  File "/tmp/.mount_FreeCAYLstxT/usr/Mod/Path/PathScripts/PathOpGui.py", line 201, in __init__
    self.form = self.getForm()  # pylint: disable=assignment-from-no-return
  File "/tmp/.mount_FreeCAYLstxT/usr/Mod/Path/PathScripts/PathFeatureExtensionsGui.py", line 230, in getForm
    form.enableExtensionsWarning.hide()
<class 'AttributeError'>: 'PySide2.QtWidgets.QWidget' object has no attribute 'enableExtensionsWarning'

OS: Ubuntu 21.04 (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.709.28846 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: c880c050c5f7601e4bae6740de71500edf2be0c7
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

luzpaz, 2022-08-28:
Can you produce with LinkDaily?

Huwyler, 2022-08-28:
Nope, with Daily, it works fine.

OS: Ubuntu 21.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

luzpaz, 2022-08-28:
@realthunder consider backporting the fix to LS3 ?