realthunder/FreeCAD_assembly3 #744: Help.... Program received signal SIGSEGV, Segmentation fault.

snelweg, 2021-07-20:
I made a mistake and now I can  not open any version of the file. How can I fix this?

`FreeCAD 0.19, Libs: 2021.717R24301 +3842 (Git)`

```
xx@DESKTOP-9FIBCI1:~/Downloads$ ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCArcIvUW
FreeCAD 0.19, Libs: 2021.717R24301 +3842 (Git)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x37840) [0x7f2872f4c840]
#1  0x7f2874651b20 in App::PropertyLink::getValue() const from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0
#2  0x7f28745e8924 in App::LinkBaseExtension::syncElementList() from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0x3f4
#3  0x7f28745e95e1 in App::LinkBaseExtension::update(App::DocumentObject*, App::Property const*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0x381
#4  0x7f28745ebbb7 in App::LinkBaseExtension::onExtendedDocumentRestored() from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0x197
#5  0x7f28744ce45a in App::DocumentObject::onDocumentRestored() from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0x8a
#6  0x7f28745a24d5 in App::GeoFeature::onDocumentRestored() from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0x95
#7  0x7f28744ac83c in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0x5ac
#8  0x7f28744ad5ec in App::Document::afterRestore(bool) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0x4c
#9  0x7f28746de71f in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0x152f
#10  0x7f28746df38b in App::Application::openDocument(char const*, bool) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0xbb
#11  0x7f28746e8f89 in App::Application::sOpenDocument(_object*, _object*, _object*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADApp.so+0xe9
#12  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(+0x16480e) [0x7f28749e280e]
#13  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(_PyObject_MakeTpCall+0x2df) [0x7f28749b1d7f]
#14  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalFrameDefault+0x51b8) [0x7f2874a44708]
#15  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(+0x196249) [0x7f2874a14249]
#16  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(_PyEval_EvalCodeWithName+0x51) [0x7f2874ab1d81]
#17  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(PyEval_EvalCodeEx+0x3a) [0x7f2874ab1dca]
#18  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(PyEval_EvalCode+0x1c) [0x7f2874ab1dec]
#19  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(+0x233e9f) [0x7f2874ab1e9f]
#20  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(+0x25838c) [0x7f2874ad638c]
#21  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libpython3.9.so.1.0(PyRun_StringFlags+0xa7) [0x7f2874ad6457]
#22  0x7f287412136c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADBase.so+0x6c
#23  0x7f28751a5c51 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADGui.so+0x91
#24  0x7f28751a5df1 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADGui.so+0x101
#25  0x7f28750b4355 in Gui::Application::open(char const*, char const*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADGui.so+0x435
#26  0x7f28751b56af in StdCmdOpen::activated(int) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADGui.so+0x72f
#27  0x7f28751ae397 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADGui.so+0x3e7
#28  0x7f287343fa18 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Core.so.5+0x6c0
#29  0x7f2873c4c928 in QAction::triggered(bool) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0x32
#30  0x7f2873c4e2b2 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0xc6
#31  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5(+0x263653) [0x7f2873d40653]
#32  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5(+0x2681a2) [0x7f2873d451a2]
#33  0x7f2873d45900 in QMenu::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0xbc
#34  0x7f2873c7c4ed in QWidget::event(QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0x133
#35  0x7f2873d46fac in QMenu::event(QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0x2d0
#36  0x7f2873c517f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#37  0x7f2873c55f5e in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0x70a
#38  0x7f287515c759 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADGui.so+0x89
#39  0x7f2873429606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Core.so.5+0x86
#40  0x7f2873c554c4 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0x286
#41  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5(+0x1aea9d) [0x7f2873c8ba9d]
#42  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5(+0x1aff7c) [0x7f2873c8cf7c]
#43  0x7f2873c517f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#44  0x7f2873c5731d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#45  0x7f287515c759 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADGui.so+0x89
#46  0x7f2873429606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Core.so.5+0x86
#47  0x7f28737cdec5 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Gui.so.5+0x677
#48  0x7f28737ce91f in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Gui.so.5+0x5f
#49  0x7f28737bbc65 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Gui.so.5+0x63
#50  /tmp/.mount_FreeCArcIvUW/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5b010) [0x7f286a466010]
#51  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f286faab23d]
#52  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/./libglib-2.0.so.0(+0x554d1) [0x7f286faab4d1]
#53  /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f286faab571]
#54  0x7f28734580e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Core.so.5+0x5e
#55  0x7f2873426019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib/libQt5Core.so.5+0x187
[... cut ...] cIvUW/usr/bin/../lib/libQt5Core.so.5+0xfd
#57  0x7f28750d5d5e in Gui::Application::runApplication() from /tmp/.mount_FreeCArcIvUW/usr/bin/../lib64/libFreeCADGui.so+0x40e
#58  /tmp/.mount_FreeCArcIvUW/usr/bin/FreeCADLink(+0x44be) [0x55cff571b4be]
#59  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f2872f3909b]
#60  /tmp/.mount_FreeCArcIvUW/usr/bin/FreeCADLink(+0x47d5) [0x55cff571b7d5]
```

[verwarming.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6847258/verwarming.zip)

snelweg, 2021-07-20:
Funny thing is it opens in Windows...

`OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3747 (Git)
Build type: Release
Branch: LinkDaily
Hash: ecbfba008c481db4cb17bf8bd514b2d12690a8e5
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Dutch/Netherlands (nl_NL)
`

closed 2021-07-21 by snelweg

snelweg, 2021-08-09:
I compiled your Linkstage3 branch myself hoping to get rid of the error and salvage the file, but I am sorry to say the file still won't open. I assume the fix is included in the latest git clone.

> OS: Debian GNU/Linux 10 (buster) (X-Cinnamon/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27928 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

`ton@DESKTOP-9FIBCI1:~/FREECAD/freecad-build/bin$ ./FreeCAD
FreeCAD 0.19, Libs: 0.19R27928 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
<PropShape> PropertyTopoShape.cpp(382): Recomputation required for document 'verwarming' on geo element version change in verwarming#Body.Shape: 1.15.70200.4 -> 1.15.70200.3.0
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x37840) [0x7f536d6e3840]
#1  0x7f5371853644 in App::PropertyLink::getValue() const from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xc
#2  0x7f53717cbd69 in App::LinkBaseExtension::syncElementList() from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x321
#3  0x7f53717ca79d in App::LinkBaseExtension::update(App::DocumentObject*, App::Property const*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x19b5
#4  0x7f53717cc7f6 in App::LinkBaseExtension::onExtendedDocumentRestored() from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x9a2
#5  0x7f537165c128 in void App::ExtensionContainer::callExtension<void (App::DocumentObjectExtension::*)()>(void (App::DocumentObjectExtension::*&&)()) const from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xf8
#6  0x7f5371657a9d in App::DocumentObject::onDocumentRestored() from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x31
#7  0x7f537178de5c in App::GeoFeature::onDocumentRestored() from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xa4
#8  0x7f5367a86e67 in App::FeaturePythonT<Part::Feature>::onDocumentRestored() from /home/ton/FREECAD/freecad-build/Mod/Part/Part.so+0x2b
#9  0x7f5371557e8d in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x60f
#10  0x7f537155769b in App::Document::afterRestore(bool) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x5d
#11  0x7f53718edb3f in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xaf7
#12  0x7f53718ec789 in App::Application::openDocument(char const*, bool) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xc5
#13  0x7f5371926ebd in App::Application::sOpenDocument(_object*, _object*, _object*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xe9
#14  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyMethodDef_RawFastCallKeywords+0x30b) [0x7f536fdc3cdb]
#15  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyCFunction_FastCallKeywords+0x25) [0x7f536fdc4a05]
#16  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x7e90) [0x7f536fbc5cb0]
#17  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7f536fced1e6]
#18  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(PyEval_EvalCodeEx+0x3e) [0x7f536fced46e]
#19  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(PyEval_EvalCode+0x1b) [0x7f536fcee23b]
#20  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(PyRun_StringFlags+0x8b) [0x7f536fcbbfeb]
#21  0x7f537081b302 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADBase.so+0xd8
#22  0x7f5372f7fcc6 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0xca
#23  0x7f5372f7f990 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x112
#24  0x7f5372dcc1a1 in Gui::Application::open(char const*, char const*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x3df
[... cut ...]
#56  0x7f536de6415b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#57  0x7f536de6c132 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#58  0x7f5372dd6e78 in Gui::Application::runApplication() from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x5f2
#59  ./FreeCAD(+0x9158) [0x56314fb6b158]
#60  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f536d6d009b]
#61  ./FreeCAD(+0x84ca) [0x56314fb6a4ca]`

luzpaz, 2021-08-09:
@snelweg please make sure to put code brackets around output so that  it doesn't auto-reference other issues in this repo.
For example:  
```
#56 0x7f536de6415b in QEventLoop::exec(QFlagsQEventLoop::ProcessEventsFlag) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#57 0x7f536de6c132 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#58 0x7f5372dd6e78 in Gui::Application::runApplication() from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x5f2
#59 ./FreeCAD(+0x9158) [0x56314fb6b158]
#60 /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f536d6d009b]
#61 ./FreeCAD(+0x84ca) [0x56314fb6a4ca]
```

as opposed to:

[image]