realthunder/FreeCAD_assembly3 #746: Assembly3 Workbench is not listed

wohltat, 2021-07-21:
I can't find the assembly3 workbench on my ubuntu machine. On my arch system (same AppImage) it is listed as expected, so i assume it is asomething about my settings. But i can't find anything in `.FreeCAD` or `.config/FreeCAD` that looks like it could cause this.
Any ideas what is going on here?

[image]

----

```
OS: Linux Mint 20.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

kevenwyld, 2021-07-21:
Workbenches are configured in `$HOME/.FreeCAD/link.user.cfg`  For example here's mine, where I have the InventorWorkbench disabled:

```
      <FCParamGroup Name="Workbenches">
        <FCText Name="Enabled">ArchWorkbench,DraftWorkbench,DrawingWorkbench,FemWorkbench,ImageWorkbench,InspectionWorkbench,MeshWorkbench,NoneWorkbench,OpenSCADWorkbench,PartDesignWorkbench,PartWorkbench,PathWorkbench,PointsWorkbench,RaytracingWorkbench,ReverseEngineeringWorkbench,RobotWorkbench,SketcherWorkbench,SpreadsheetWorkbench,StartWorkbench,SurfaceWorkbench,TechDrawWorkbench,TestWorkbench,WebWorkbench,Assembly3Workbench,GliderWorkbench,FastenersWorkbench,BIMWorkbench,A2plusWorkbench,Assembly4Workbench,CurvesWorkbench,ThreadProfileWorkbench,</FCText>
        <FCText Name="Disabled">InventorWorkbench,</FCText>
      </FCParamGroup>
```

You can also see disabled workbenches in the `Customize` control panel.  Tools -> Customize -> Workbenches tab

[image]

Do you see anything in either of those places?

wohltat, 2021-07-22:
I have found it. It was not in the `.FreeCAD/link.user.cfg`.

The culprit was in the .FreeCAD/Mod folder in an other workbench. It was the pyrate workbench. I reinstalled it and i can see the assembly3 workbench now.

But since this is rather strange i looked further into this. The difference that caused this was **an empty `__init__.py` file in the pyrate/freecad folder**. After replacing it with the new (non-empty) version it works. Also if i empty the `__init__.py` in the newly installed pyrate workbench, the assembly3 workbench disappears again.
Further in the working state, there is additional stuff printed in the console on FreeCAD startup from other workbenches. Seems as if this problem interrupts workbench loading and causes other workbenches not to load.
In another folder there is an empty `__init__.py` as well but that one doesn't cause any trouble.

luzpaz, 2021-07-22:
@wohltat Do you mind opening a ticket on https://salsa.debian.org/mess42/pyrate/ and letting them know about this?  
We should also think about running a test for this to avoid issues like this in the future.

cc @joha2 (associated with pyrate)

wohltat, 2021-07-23:
@luzpaz Isn't this the cause for the issue more on the FreeCAD side? Since other packages also have empty `__init__.py`s and it already working with the latest  version of pyrate.
Running a test to capture these cases is a good idea. Or maybe trying to find why this interferes at all with other workbench loading.

luzpaz, 2021-07-23:
yea.. good point. can you open a ticket on the FreeCAD bugtracker for this and x-post it back here?

realthunder, 2021-07-23:
> But since this is rather strange i looked further into this. The difference that caused this was **an empty `__init__.py` file in the pyrate/freecad folder**. After replacing it with the new (non-empty) version it works.

I just checked pyrate source. They do have a proper [__init__.py](https://salsa.debian.org/mess42/pyrate/-/blob/master/freecad/__init__.py) in its `freecad` directory. So how did you get an empty `__init__.py` in the first place? The code in `__init__.py` is to add that directory as a search path for the `freecad` package. Without that line, the `freecad` package path will be cleared and set exclusively to this directory. So, depending on the position of this rouge package, all sub-package before it that is also in `freecad` namespace will be hidden, and those that follows it can still be found.

wohltat, 2021-07-23:
@realthunder This is what i was saying. The current version is fine and updating solved the problem. It was an older version (april 2018 or earlier) that had an empty `__init__.py` and broke things.

empty `__init__.py`s seem to be quite common
```
>>> find -size 0 -name __init__.py   
./pyrate/freecad/PyrateWorkbench/__init__.py
./pyrate/freecad/PyrateWorkbench/Gui/Resources/__init__.py
./pyrate/freecad/PyrateWorkbench/Gui/Resources/icons/__init__.py
./pyrate/freecad/PyrateWorkbench/Gui/__init__.py
./pyrate/pyrateoptics/core/__init__.py
./pyrate/pyrateoptics/io/__init__.py
./pyrate/pyrateoptics/analysis/__init__.py
./pyrate/pyrateoptics/sampling2d/__init__.py
./pyrate/pyrateoptics/optimize/__init__.py
./pyrate/pyrateoptics/material/__init__.py
./pyrate/demos/__init__.py
./pyrate/tests/__init__.py
./cqparts/src/cqparts_fasteners/solidtypes/__init__.py
./cqparts/src/cqparts_fasteners/fasteners/__init__.py
./cqparts/src/cqparts_toys/train/__init__.py
./cqparts/tests/t_cqparts_fasteners/__init__.py
./cqparts/tests/t_cqparts/__init__.py
./cqparts/tests/t_cqparts_template/__init__.py
./Render/renderers/__init__.py
./workfeature/Resources/__init__.py
./workfeature/Resources/Icons/__init__.py
./workfeature/Utils/__init__.py
./BOLTSFC/BOLTS/freecad/flanges/__init__.py
./BOLTSFC/BOLTS/freecad/profile_l/__init__.py
./BOLTSFC/BOLTS/freecad/hex_socket/__init__.py
./BOLTSFC/BOLTS/freecad/profile_i/__init__.py
./BOLTSFC/BOLTS/freecad/bearings/__init__.py
./BOLTSFC/BOLTS/freecad/washer/__init__.py
./BOLTSFC/BOLTS/freecad/__init__.py
./BOLTSFC/BOLTS/freecad/profile_hollow/__init__.py
./BOLTSFC/BOLTS/freecad/nut/__init__.py
./BOLTSFC/BOLTS/freecad/hex/__init__.py
./BOLTSFC/BOLTS/freecad/connecting_bars/__init__.py
./BOLTSFC/BOLTS/freecad/batteries/__init__.py
./BOLTSFC/BOLTS/freecad/extrusions/__init__.py
./BOLTSFC/BOLTS/freecad/pipes/__init__.py
./BOLTSFC/BOLTS/freecad/profile_c/__init__.py
./BOLTSFC/BOLTS/freecad/profile_z/__init__.py
./BOLTSFC/BOLTS/bolttools/__init__.py
./BOLTSFC/BOLTS/gui/__init__.py
./LCInterlocking/panel/__init__.py
./LCInterlocking/lasercut/__init__.py
./A2plus/__init__.py
```

realthunder, 2021-07-24:
The `__init__.py` that matter are the ones in `freecad` directory. `freecad` is the name chosen as the `namespace` package for holding all new style FreeCAD external modules.

wohltat, 2021-07-24:
Ok, so then there is only one left from additional the workbenches 

`./BOLTSFC/BOLTS/freecad/__init__.py`

But this one doesn't seem to cause problems.

realthunder, 2021-07-24:
It depends on the order FreeCAD load those modules. Any modules under `freecad` namespace loaded before this rouge module will have problem.

luzpaz, 2021-09-12:
> Ok, so then there is only one left from additional the workbenches
> 
> `./BOLTSFC/BOLTS/freecad/__init__.py`
> 
> But this one doesn't seem to cause problems.

@berndhahnebach looks like BOLTS `__init__.py` may be causing some issues

luzpaz, 2022-04-08:
Is this still an issue?

closed 2022-04-18 by luzpaz