realthunder/FreeCAD_assembly3 #747: Draft input problems in LinkDaily

carlopav, 2021-07-23:
Hello, found an annoying bug in LinkDaily: Draft input tools do not work.

`OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.716.24301 +3839 (Git)
Build type: Release
Branch: LinkDaily
Hash: 31965a2cdaa4a5870b9608ad58c25115be334f13
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Italian/Italy (it_IT)
`
Steps to reproduce:
1. Switch to Draft WB
2. Draft Grid correctly displays
3. Activate a command, for example Draft_Line
4. The task panel is correctly displayed, waiting for user to input first point
5. Problem: when you move the mouse in the 3d view, the coordinates are not updated in the task panel and you cannot input a point by clicking in the 3d view. The task panel input works instead.

thanks for your attention and have a good day!

realthunder, 2021-07-23:
The problem has been fixed in 0717 release. Please download and try.

carlopav, 2021-07-23:
thx, i Will test next week!

carlopav, 2021-08-01:
Tested with:

`OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Italian/Italy (it_IT)
`
Draft tools work good now, thanks!

I still experience some problems with the mouse cursors: in current master, the mouse cursor is displayed as a cross and have the current active snap icon near it, while in LinkDaily it has the standard arrow without any snap marker.

closed 2022-07-11 by luzpaz