realthunder/FreeCAD_assembly3 #748: Issues with pad fuction

mwlandolsi, 2021-07-24:
Hello Every body.

1- infinte pad : 

I got "infinite" pad when the type of limit is set to "to first" (selected limit face is circular) or "to first".

[image]

2- Choice of limit is not obvious

[image]

----------------------------------
OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
[pad issues 07-2021.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6872234/pad.issues.07-2021.zip)

realthunder, 2021-07-25:
I have added a new option to make this work. You can uncheck the option if you want to pad/pocket up to some concave face(s).

[image]

mwlandolsi, 2021-07-25:
Yes! Great! Thanks for your work. I will test it when a new release is published. Many thanks. I hope that I could help in coding!

mwlandolsi, 2021-09-11:
Yes many thanks, this will solve many design problems and make FreeCAD  Link stage more competitive especially with plastic part design. this is very useful.
Many thanks Solved with 11.09.2021 release.

mwlandolsi, 2022-01-19:
Looks like this feature is not any more available in 11012022 daily

closed 2022-02-23 by mwlandolsi