realthunder/FreeCAD_assembly3 #750: Strange behaviour in Part Design chamfer

WiseroOrb, 2021-07-24:
The result of chamfer produces some bulges, but the wireframe is OKish

[image]
[image]
[image]
[image]

OS: Ubuntu 21.04 (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: Italian/Italy (it_IT)

mwlandolsi, 2021-07-25:
I confirm this issue but got a different strange edge !
[image]
[fillet with draft.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6874670/fillet.with.draft.FCStd.zip)

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-07-27:
I think this is an OpenCADCADE problem. It simply can't handle this type of fillet/chamfer. The fillet shape around joint of the slope and the vertical edge is non cylindrical.  The best looking shape I can get is the move the second pad after the fillet and add another fillet below. As you can see there is still an extra edge there, i.e. the surface is not smooth.

Another way to achieve the same result is to create a `New solid` for each pad, and do fillet on them, and then use `Boolean` feature to merge them. It seems that `Chamfer` only works using this way on my computer. 

https://user-images.githubusercontent.com/1207888/127126361-51ff2697-4014-4bd2-960b-af0dceb15ac5.mp4

realthunder, 2021-07-27:
BTW, the chamfer size will be slightly different. The slope edge having 0.95mm, which is about the same for a 1mm chamfer on vertical edge.

[image]

WiseroOrb, 2021-07-28:
Fair enough, chamfer is quite buggy.
However I've found that you can replicate the functionality of the chamfer with SubtractivePipe and a triangle shape.
Doing the chamfer in this way gives always perfect results.
Do you think is possible to automate this process? Maybe replacing the current chamfer with this procedure?
I know there are issues, but it could be a way to finally solve the chamfers in FC
[image]
[image]
[image]
[image]