realthunder/FreeCAD_assembly3 #751: A "cut" feature gave me PAD feature !

mwlandolsi, 2021-07-24:
Hello every body, 
This part is supposed  to be with holes,
[image]
[image]
[monture.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6872982/monture.FCStd.zip)

luzpaz, 2021-07-24:
Please always add you full [About](https://wiki.freecadweb.org/About) info

mwlandolsi, 2021-07-24:
Sorry,
I always add it, but this time i missed it :)
OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-07-25:
The problem happens because the `Refine` operation sometimes reverses the solid. I have just fixed that. The first feature that suffers from this problem is `Pocket`. If you turn off `Two side` rendering, you may discover this problem sooner, with the shape having black faces. Note that, once the problem is fixed, the shape got reversed back, and some your sketch mapping offset may have the wrong orientation. You'll need to adjust that.

[image]

[image]

You can turn off two side rendering for all new objects in preference

[image]

mwlandolsi, 2021-07-25:
thank you for your answer. pleased to get from you. Waiting next release.

mwlandolsi, 2021-09-11:
Many thanks, looks good now.

closed 2021-09-11 by mwlandolsi