realthunder/FreeCAD_assembly3 #752: mac CONDA July 17 release crashes when mesh selected

sslupsky, 2021-07-26:
The latest release (July 17, 2021) CONDA build crashes when a mesh is selected in the model view.  The same mesh selected using the Big Sur build does not crash the app.  However, issue #582 is still present in the Big Sur build which causes the app to crash if you attempt to create a mesh.

```
Process:               FreeCAD [2647]
Path:                  /Applications/FreeCAD LS3 Conda.app/Contents/Resources/bin/FreeCAD
Identifier:            ???
Version:               ??? (2021)
Code Type:             X86-64 (Native)
Parent Process:        ??? [2639]
User ID:               501

Date/Time:             2021-07-26 13:34:38.463 -0600
OS Version:            macOS 11.4 (20F71)
Report Version:        12

Time Awake Since Boot: 20000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [2647]

VM Regions Near 0x10:
--> 
    __TEXT                      1033d9000-1033e0000    [   28K] r-x/r-x SM=COW  /Applications/FreeCAD LS3 Conda.app/Contents/Resources/bin/FreeCAD

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libCoin.80.0.1.dylib          	0x0000000105ff6e10 SoCallbackAction::~SoCallbackAction() + 80
1   libCoin.80.0.1.dylib          	0x0000000105ff6fbe SoCallbackAction::~SoCallbackAction() + 14
2   libFreeCADGui.dylib           	0x0000000103795e74 SoFCRenderCacheManagerP::initAction() + 68
3   libFreeCADGui.dylib           	0x000000010379a25b SoFCRenderCacheManager::render(SoGLRenderAction*) + 395
4   libFreeCADGui.dylib           	0x00000001037cf040 Gui::SoFCUnifiedSelection::GLRenderBelowPath(SoGLRenderAction*) + 160
5   libCoin.80.0.1.dylib          	0x000000010624d41a SoSeparator::GLRenderBelowPath(SoGLRenderAction*) + 586
6   libCoin.80.0.1.dylib          	0x0000000105ff5179 SoAction::beginTraversal(SoNode*) + 105
7   libCoin.80.0.1.dylib          	0x0000000105ff8dc4 SoGLRenderAction::beginTraversal(SoNode*) + 708
8   libFreeCADGui.dylib           	0x00000001037e573e Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) + 1534
9   libCoin.80.0.1.dylib          	0x0000000105ffb303 SoGLRenderActionP::renderSingle(SoNode*) + 371
10  libCoin.80.0.1.dylib          	0x0000000105ff8ffc SoGLRenderActionP::render(SoNode*) + 412
11  libCoin.80.0.1.dylib          	0x0000000105ff8d97 SoGLRenderAction::beginTraversal(SoNode*) + 663
12  libFreeCADGui.dylib           	0x00000001037e573e Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) + 1534
13  libCoin.80.0.1.dylib          	0x0000000105ff371a SoAction::apply(SoNode*) + 362
14  libFreeCADGui.dylib           	0x00000001037e455a Gui::SoBoxSelectionRenderAction::apply(SoNode*) + 42
15  libCoin.80.0.1.dylib          	0x0000000106317cb7 SoRenderManager::renderScene(SoGLRenderAction*, SoNode*, unsigned int) + 231
16  libCoin.80.0.1.dylib          	0x0000000106317b2b SoRenderManager::actuallyRender(SoGLRenderAction*, int, int, int) + 155
17  libCoin.80.0.1.dylib          	0x000000010631772d SoRenderManager::renderSingle(SoGLRenderAction*, int, int, int) + 973
18  libCoin.80.0.1.dylib          	0x00000001063172e3 SoRenderManager::render(SoGLRenderAction*, int, int, int) + 211
19  libFreeCADGui.dylib           	0x00000001038b8361 Gui::View3DInventorViewer::renderScene() + 705
20  libFreeCADGui.dylib           	0x0000000103873691 SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) + 305
21  libFreeCADGui.dylib           	0x000000010387add2 SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) + 66
22  libQt5Widgets.5.12.9.dylib    	0x0000000105261552 QWidget::event(QEvent*) + 1202
23  libQt5Widgets.5.12.9.dylib    	0x00000001052cffe3 QFrame::event(QEvent*) + 37

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x00007f891dfe0400  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x00007ffeec822e98
  rdi: 0x0000000000000000  rsi: 0x0000000000000001  rbp: 0x00007ffeec822dd0  rsp: 0x00007ffeec822da0
   r8: 0x000000000000a5b0   r9: 0x00007f88ac4a9630  r10: 0x00000000fffcffff  r11: 0x00007f88ac4acc40
  r12: 0x0000000000000145  r13: 0x00007f88cda52340  r14: 0x00007f88ac421540  r15: 0x00007f88cda522b0
  rip: 0x0000000105ff6e10  rfl: 0x0000000000010246  cr2: 0x0000000000000010
  
Logical CPU:     4
Error Code:      0x00000004 (no mapping for user data read)
Trap Number:     14

Thread 0 instruction stream:
  00 c7 00 01 00 00 00 5d-c3 0f 1f 80 00 00 00 00  .......]........
  55 48 89 e5 41 57 41 56-41 55 41 54 53 50 48 8d  UH..AWAVAUATSPH.
  05 bb 63 4e 00 48 89 07-4c 8d bf d0 00 00 00 48  ..cN.H..L......H
  89 7d d0 4c 8b af d0 00-00 00 45 8b 65 34 45 85  .}.L......E.e4E.
  e4 7e 39 31 db 66 2e 0f-1f 84 00 00 00 00 00 90  .~91.f..........
  49 8b 45 38 48 8b 3c d8-0f 1f 84 00 00 00 00 00  I.E8H.<.........
 [4c]8b 77 10 e8 55 ba 43-00 4c 89 f7 4d 85 f6 75  L.w..U.C.L..M..u	<==
  ef 48 ff c3 4c 39 e3 75-d7 4d 8b 2f 45 8b 65 64  .H..L9.u.M./E.ed
  45 85 e4 7e 37 45 31 f6-0f 1f 84 00 00 00 00 00  E..~7E1.........
  49 8b 45 68 4a 8b 3c f0-0f 1f 84 00 00 00 00 00  I.EhJ.<.........
  48 8b 5f 10 e8 15 ba 43-00 48 89 df 48 85 db 75  H._....C.H..H..u
  ef 49 ff c6 4d 39 e6 75-d7 4d 8b 2f 45 8b a5 a4  .I..M9.u.M./E...
  
Thread 0 last branch register state not available.

Binary Images:
       0x1033d9000 -        0x1033dffff +FreeCAD (0) <430ABBB5-5A9C-3285-AEA3-D84FA5CB2ABF> /Applications/FreeCAD LS3 Conda.app/Contents/Resources/bin/FreeCAD
       0x1033e6000 -        0x1040defff +libFreeCADGui.dylib (0) <84765983-0D7A-3A61-AB73-3CDB29C4A837> /Applications/FreeCAD LS3 Conda.app/Contents/Resources/lib/libFreeCADGui.dylib
       0x1045cf000 -        0x104931fff +libpython3.9.dylib (0) <B7915BD8-43CA-3C5B-B3CA-674A201054DE> /Applications/FreeCAD LS3 Conda.app/Contents/Resources/lib/libpython3.9.dylib
       0x1049f1000 -        0x104e0cfff +libFreeCADApp.dylib (0) <8005E51A-5C2F-391B-BC6C-9852E66B85B5> /Applications/FreeCAD LS3 Conda.app/Contents/Resources/lib/libFreeCADApp.dylib
       0x105043000 -        0x10517cfff +libFreeCADBase.dylib (0) <B8887521-8F0A-335C-9F3A-624E512E5ACE> 
```

sslupsky, 2021-07-26:
I've attached a copy of the file here.
[tr12 collar.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6880832/tr12.collar.FCStd.zip)

thomasgi1, 2021-08-07:
Crashes here too with exactly the same stack trace, but is was not required to select the mesh. Just hovering the mouse over the mesh item in the tree view caused the crash here.

OS: macOS Mojave (10.14)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

realthunder, 2022-01-10:
Fixed by https://github.com/realthunder/coin/commit/41dd4382f6068dd824dd2f515b087bebeeeb066c

closed 2022-01-10 by realthunder