realthunder/FreeCAD_assembly3 #753: asm3 says redundant constraints

kth001, 2021-07-27:
In the attached file, I built an assembly that asm3 says it has redundant constraints,
but I think there is no redundant constraints. 
I changed all the plane coincident constraints to axial alignment constraints, but it didn't helped.
Did I missed something? or it's the bug of this workbench? please let me know
[assemblytest.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6883101/assemblytest.zip)

### freecad info
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Korean/South Korea (ko_KR)

### assembly 3 info
Branch: Master

realthunder, 2021-07-27:
`PlaneCoincident` removes 5 DOF, leaving only a rotational DOF. The last `PlaneCoincident` in your assembly causes the redundancy, because the levers can only move in a plane, i.e they only have 2 tranlational DOF and 1 rotational DOF left. You can replace the last `PlaneCoincident` with a `PointCoincident`, and adding a third face element for specifying the workplane to make this a 2D `PointCoincident`.

kth001, 2021-07-27:
thanks for explaining. This problem was due to my lack of understanding about DOF.

closed 2021-07-27 by kth001