realthunder/FreeCAD_assembly3 #754: mac conda daily 0717 crashes in sketcher, no geometry needed

bozzzzo, 2021-07-27:
- Preferences -> General -> General -> Start up -> Auto load module after startup: Start
- Resart freecad
- New document
- sketcher
- xy plane
- close sketch -> crash
[backtrace.txt](https://github.com/realthunder/FreeCAD_assembly3/files/6888165/backtrace.txt)

bozzzzo, 2021-07-27:
When the initial workbench is sketcher the above sequence does not crash.

thomasgi1, 2021-08-07:
NO crash here:

OS: macOS Mojave (10.14)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

ruchg, 2021-08-11:
Been using FreeCAD installed off package manager on Fedora 34 and Cinnamon

Downloaded AppImage and made executable

Was having difficulties, similar to this, so I configured the Preferences like above and now the AppImage always core dumps.

`
[ruchg@localhost Downloads]$ ./FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCA0nuIbK
FreeCAD 0.19, Libs: 2021.717R24301 +3627 (Git)
Sheet Metal workbench loaded
libGL error: MESA-LOADER: failed to open nouveau: /tmp/.mount_FreeCA0nuIbK/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by /usr/lib64/dri/nouveau_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: nouveau
libGL error: MESA-LOADER: failed to open nouveau: /tmp/.mount_FreeCA0nuIbK/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by /usr/lib64/dri/nouveau_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: nouveau
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCA0nuIbK/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by /usr/lib64/dri/swrast_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCA0nuIbK/AppRun: line 46:  3525 Aborted                ###  **(core dumped)** "$PREFIX/bin/$name" "$@"
[ruchg@localhost Downloads]$ ./FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCAKmk9Nt
FreeCAD 0.19, Libs: 2021.717R24301 +3627 (Git)
Sheet Metal workbench loaded
libGL error: MESA-LOADER: failed to open nouveau: /tmp/.mount_FreeCAKmk9Nt/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by /usr/lib64/dri/nouveau_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: nouveau
libGL error: MESA-LOADER: failed to open nouveau: /tmp/.mount_FreeCAKmk9Nt/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by /usr/lib64/dri/nouveau_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: nouveau
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAKmk9Nt/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by /usr/lib64/dri/swrast_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCAKmk9Nt/AppRun: line 46:  3785 Aborted                 ### **(core dumped)** "$PREFIX/bin/$name" "$@"
[ruchg@localhost Downloads]$ ./FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCAldyMXQ
FreeCAD 0.19, Libs: 2021.717R24301 +3627 (Git)
Sheet Metal workbench loaded
libGL error: MESA-LOADER: failed to open nouveau: /tmp/.mount_FreeCAldyMXQ/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by /usr/lib64/dri/nouveau_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: nouveau
libGL error: MESA-LOADER: failed to open nouveau: /tmp/.mount_FreeCAldyMXQ/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by /usr/lib64/dri/nouveau_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: nouveau
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAldyMXQ/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.29 not found (required by /usr/lib64/dri/swrast_dri.so) (search paths /usr/lib64/dri)
libGL error: failed to load driver: swrast
/tmp/.mount_FreeCAldyMXQ/AppRun: line 46:  3850 Aborted                 ### **(core dumped)** "$PREFIX/bin/$name" "$@"
[ruchg@localhost Downloads]$ '

Regards