realthunder/FreeCAD_assembly3 #755: [bug] Additive pipe preview OK but model not

Miksu80, 2021-07-28:
Issue with attached model, when doing additive pipe, it doesnt show up in model but preview OK. When mirroring the object, again preview OK, but model not. Propably not link related, havent tested on non-link-versions.

[image]
[image]
[image]
[image]

[Coolingduct_debug.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6890429/Coolingduct_debug.zip)

mwlandolsi, 2021-07-28:
works fine for me on Linux mint.
[image]
[Coolingduct_debug-.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6891679/Coolingduct_debug-.FCStd.zip)

OS: KDE Flatpak runtime (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.15.3
Coin version: 4.0.0
OCC version: 7.3.0
Locale: English/United States (en_US)

Miksu80, 2021-07-28:
@mwlandolsi looks like U have a older version, it might have something to do with it. I've had some situations like this but was able to recover by changing the params etc. but this one just keeps hidden to me.. Mine version likes to cross the inner and outer faces so they go to 0 thickness at some point but your looks nice with thick walls.. Can you share your file when its looking ok so I can check what mine says on that?

mwlandolsi, 2021-07-28:
Yes, I tested it on 0.19.2
but on Branch: LinkDaily I got error!
the best result I got by activation "New solid" Option
[image]

but when I click On OK, I got just the pipe without mirror. You may be need to edit mirror again.
after editing the mirror and choosing also "new solid " option, I got the mirrored shape. You must keep the features : AdditivePipe and the mirror (Mirrored an Mirrored001) set to visible.
You can notice that the preview (blue) is different from the solid when in editing pipe.
this is the final result, but with some display errors (red arrows on the 3d). It could be an OCC problem. May be the developers could confirm this or not.

[image]

By the way, I want to pay attention to my reported issues #597 and #624 which, also, are related to pipe and sweep functions.

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)