realthunder/FreeCAD_assembly3 #756: Can't change enumeration property after setting up a configuration table

alexneufeld, 2021-07-28:
I *think* I've managed to find a weird issue with App::PropertyEnumeration and configuration tables.
 steps to produce the behaviour:
- Open a new document
- Add an object (App::Part) and assign it an enum property
- set the enum to some value, e.g. [<<1>>, <<2>>, <<3>>] (right click on the "Enum" dropdown and set expression)
- Add a spreadsheet to the document and create a simple configuration table that points to a property in the same object as we added the previous enum property to ( eg: Part.ConfigProperty )
- now try to change the enumeration of the first property we created. (right click on its "Enum" drop-down and set the expression to [<<4>>, <<5>>, <<6>>])
- The first enumeration property will fail to update. It gets completely stuck and can't be changed.

[image]

Tested using FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage

OS: Manjaro Linux (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/Canada (en_CA)

[enum_bug_demo.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6890917/enum_bug_demo.zip)

realthunder, 2021-07-31:
That's indeed a quite obscure bug. Thanks for reporting!

alexneufeld, 2021-08-01:
Thank you for taking a look at it!
I'll have to wait for the next LinkDaily release to test the changes. Will reopen if the issue is still present there.

closed 2021-08-01 by alexneufeld