realthunder/FreeCAD_assembly3 #757: [Feature request] 3D-sketches to part design -WB

Miksu80, 2021-07-28:
It'd be nice to have some sort of 3D-sketches so you can do plumbing, cabling, ducting etc. easier. When for example designing printer cooling, the sweep offers tools but sketching is difficult because you've got to maneuver in tight space and the duct should be quite fluent. 

At simplest it could be a point-in-space based lines and ability to make corners round. Even if there would not be constraints, if the points could be moved wit the triad, it'd make a superb tool for many things.

It could work so that when you create a 3D-sketch, 1st point would be automatically created and placed to [0,0,0]. Then it'd had the triad so you could drag it simply where you need it. Then there'd be "add point" and it'd create 2nd point to same place as the previous and a line between. Then the 2nd point would be active and you could drag it with triad. After 2nd vertex, when "add point" would be pressed, it'd behave similarly as previous. At any time, you could select a different corner point and drag it with triad. When a point is activated and it is connected with 2 lines, you could have the option to "add fillet" and it'd ask for the radius for it. You could also select many points with ctrl-click or selection rectangle and add the fillet to multiple points. When these are not constrained to anything, it keeps together easily and doesn't jump around but editing would be an easy task with the triads.

[image]

bennvan, 2021-07-29:
I second this. It would also be useful in creating complex surfaces if you could pierce the points onto other wires etc.

Miksu80, 2021-07-30:
Some idea-flow, it doesn't necessarily make any sense 😄 :

It could work so that each 3D-sketch has a sheet below it and it could also be edited directly, sheet would have every point on its own line. Columns would be at least Point number, X, Y, Z, Point(s) connected with line, Radius etc. like that:

[image]

When you have the table, it'd give you the option to mass-fill R's and create simple shapes by hand. It would also give you the bending table for tube bending machine directly. When moved the points by hand, it would update the positions to table and when edited the table by hand, it'd update the 3D. There could be an button "make loop" to connect the 1st and last point for surface making.

At some point there could be possibility to use reference points from 3D-model. Then the table could have the "external" points included but with different index designator for filtering and differenting them from sketch points. Sketch points could have transformation dependency information in the table (reference point index, transformation [X,Y,Z] or [direction, length]).

mads-leth-danielsen, 2021-08-01:
you can make 3d sketch in freecad but it is difficult to edit see
https://youtu.be/-VaQqvHBT2Q

Miksu80, 2021-08-02:
I know that it is possible to do like that but it's hard to do any real 3d sketching that way.

bambukouk, 2021-08-30:
+1 on that enhancement,
I do a lot of piping in my designs and at the moment it is right PITA in FreeCAD.
Thank you!