realthunder/FreeCAD_assembly3 #758: [Bug] Crash when invoking part design feature wrap

bennvan, 2021-07-29:
Crashes when invoking seemingly any tool that would normally be placed in a part design feature wrap. This does not seem to occur on the current stable build.

Steps to reproduce:
1. Create new part design body
2. Draw a sketch
3. Use offset 2D tool from part wb on sketch
4. Select yes to feature wrap if not auto-enabled
5. Crashes

OS: Ubuntu 20.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/Australia (en_AU)

luzpaz, 2021-08-03:
@realthunder just wanting to make sure (apologies if you already know this) but you know you can auto-close tickets via commit summary? Example, adding: `Closes #758` will autoclose the ticket one the commit is merged.

realthunder, 2021-08-03:
Yeah, I used to do that. But there were user complaining about not able to find their issues. You know because the default filter is to show only open issue. Considering the growing speed of the queue, I guess I'll start closing using comment again.

luzpaz, 2021-08-03:
@realthunder  

All they have to do is search their issues:
[image]

but of course some users aren't aware of that. In that case: you could ping them in the commit like:
`Fixes #758 reported by @bennvan`

Maybe that's a solution?

Shkolik, 2021-08-03:
> I'll start closing using comment again.

Please do not do it! Better close some old and already fixed issues. Sometime it's helpful to see is issue was tested by someone else than you. 

But if you can ping user who submitted issue it will be useful as well - he can test it quicker and close by himself.

Thanks!

realthunder, 2021-08-03:
Oh, I didn't know about the pinging. That's nice!

luzpaz, 2021-08-04:
@Shkolik pinging is nice but users forget or neglect to follow through (I know because I do this to, unfortunately). 

> Better close some old and already fixed issues.

Disagree, the issue queue becomes unmanageable and overwhelming. 

I think it's better to remotely close and ping. If the issue arises, user can re-open ticket or devs can search and find close tickets.

bennvan, 2021-08-04:
> All they have to do is search their issues

Couldn't agree more. I am just grateful these get looked at and fixed so fast in the first place. For an issue like this, it's not like it needs a second opinion anyway.
Overall I think maintenance of your own repo should be at your discretion. Whatever makes workflow better/more manageable.

luzpaz, 2021-08-04:
@bennvan agreed.  
The ideal workaround would be that github would allow the dev to assign a tag to the ticket being referenced, triggered by the committed fix. But alas...

luzpaz, 2022-01-12:
@bennvan please retest on latest release

closed 2022-01-12 by bennvan