realthunder/FreeCAD_assembly3 #760: [Bug] Pocket from face works in preview but not when applied

Miksu80, 2021-07-30:
Pocket from face, preview OK:
[image]

Formed shape something weird:
[image]

File:
[pocket_from_face_bug.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6907013/pocket_from_face_bug.zip)

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

realthunder, 2021-07-31:
Not sure about the cause, but if you manually recompute `Sketch035` (right click -> Recompute object), and then do a document recompute can fix the problem. Probably some obscured bug hidden somewhere.

Miksu80, 2021-08-03:
Thank you. Hard to tell where the problem is -> I'll close this

closed 2021-08-03 by Miksu80