realthunder/FreeCAD_assembly3 #761: Incorrect visualization of the sketch in the model

IbnAlex, 2021-07-30:
On the model in the correct planes. But if you hover the cursor over the view in the construction tree, they are shown in other planes rotated 90°. This is not the case in FreeCAD 0.20.

https://user-images.githubusercontent.com/82962107/127707232-f1ced9e9-6174-40f4-a3de-54c79724a996.mp4

OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Russian/Russia (ru_RU)

realthunder, 2021-07-31:
There is a recent [fix](https://github.com/realthunder/FreeCAD/commit/f3c3545c2c38afc8639c01e2751d8c0d951fbb1a) for this. But I am not sure. Can you please post the file with the problem?

IbnAlex, 2021-07-31:
Yes, of course.

IbnAlex, 2021-07-31:
Yes, of course.
In FreeCAD 0.20 sketches are displayed correctly. Link to file:
https://disk.yandex.ru/d/0N_YGi5g3vwm_A

IbnAlex, 2021-07-31:
Sorry, I accidentally closed my application. I'm a newbie.

realthunder, 2021-08-01:
Yep, it's confirmed fixed.

IbnAlex, 2021-08-01:
Okay.
But for now, this problem is still present in the "FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage" version. Will the fix appear in the new update? Or is it planned? Or is it already fixed in the stable version - Stable?

IbnAlex, 2021-08-01:
Important addition. This problem occurs under the Windows 7 operating system.
There is no such error when working under Linux.

realthunder, 2021-08-02:
That's strange. I just tried 0717 Windows release and cannot reproduce the error. Can you please post your `link.user.cfg` file in `c:\users\<username>\AppData\Roaming\FreeCAD\link.user.cfg`

IbnAlex, 2021-08-03:
Okay.
Here's the link.user.txt file. txt - change to cfg, or download from: https://disk.yandex.ru/d/DdmGeD6rbz9uKg
link.user.txt](https://github.com/realthunder/FreeCAD_assembly3/files/6926349/link.user.txt)

luzpaz, 2022-07-11:
@IbnAlex please revisit this with the latest version