realthunder/FreeCAD_assembly3 #762: Crash, when transforming a sub-object link to a sketch

thomasgi1, 2021-07-31:
Steps to reproduce the crash:

- start FC
- change to PD workbench
- create a sketch in XY plane
- draw a circle
- close the sketch
- create a sub-object link to that sketch
- open the transform tool for the created link and move it
- close the transform tool

=> crash:
Process:               FreeCAD [64641]
Path:                  /Applications/FreeCAD asm3.app/Contents/Resources/bin/FreeCAD
Identifier:            FreeCAD
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [64634]
Responsible:           FreeCAD [64641]
User ID:               501

Date/Time:             2021-07-31 08:52:06.217 +0200
OS Version:            Mac OS X 10.14.6 (18G9216)
Report Version:        12
Anonymous UUID:        00D0AAE4-4FA2-1AA2-1DC3-429829F4143B

Sleep/Wake UUID:       07BC8E11-75F5-4C78-97CF-64273B730A69

Time Awake Since Boot: 1000000 seconds
Time Since Wake:       2200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [64641]

VM Regions Near 0x10:
--> 
    __TEXT                 000000010458e000-0000000104595000 [   28K] r-x/r-x SM=COW  /Applications/FreeCAD asm3.app/Contents/Resources/bin/FreeCAD

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libCoin.80.dylib              	0x00000001071b8e50 SoCallbackAction::~SoCallbackAction() + 144
1   libCoin.80.dylib              	0x00000001071b8fbe SoCallbackAction::~SoCallbackAction() + 14
2   libCoin.80.dylib              	0x0000000107232d12 SoDragger::~SoDragger() + 194
3   libFreeCADGui.dylib           	0x00000001049a9dce Gui::TDragger::~TDragger() + 14
4   libCoin.80.dylib              	0x000000010739580c SoBase::destroy() + 172
5   libCoin.80.dylib              	0x0000000107388f1a SoBaseList::truncate(int) + 74
6   libCoin.80.dylib              	0x0000000107399d01 SoChildList::~SoChildList() + 17
7   libCoin.80.dylib              	0x00000001073f0ec5 SoGroup::~SoGroup() + 37
8   libCoin.80.dylib              	0x000000010740eb4e SoSeparator::~SoSeparator() + 14
9   libCoin.80.dylib              	0x000000010739580c SoBase::destroy() + 172
10  libCoin.80.dylib              	0x0000000107388f1a SoBaseList::truncate(int) + 74
11  libCoin.80.dylib              	0x0000000107399d01 SoChildList::~SoChildList() + 17
12  libCoin.80.dylib              	0x00000001073f0ec5 SoGroup::~SoGroup() + 37
13  libCoin.80.dylib              	0x0000000107412e18 SoSwitch::~SoSwitch() + 56
14  libCoin.80.dylib              	0x000000010739580c SoBase::destroy() + 172
15  libCoin.80.dylib              	0x0000000107389058 SoBaseList::remove(int) + 40
16  libCoin.80.dylib              	0x00000001073f13d7 SoGroup::removeChild(int) + 71
17  libFreeCADGui.dylib           	0x0000000104aac59f Gui::coinRemoveAllChildren(SoGroup*) + 735
18  libFreeCADGui.dylib           	0x000000010498d39b Gui::SoFCPathAnnotation::~SoFCPathAnnotation() + 27
19  libFreeCADGui.dylib           	0x000000010498d62e Gui::SoFCPathAnnotation::~SoFCPathAnnotation() + 14
20  libCoin.80.dylib              	0x000000010739580c SoBase::destroy() + 172
21  libCoin.80.dylib              	0x0000000107388f1a SoBaseList::truncate(int) + 74
22  libCoin.80.dylib              	0x0000000107399d01 SoChildList::~SoChildList() + 17
23  libCoin.80.dylib              	0x00000001073f0ec5 SoGroup::~SoGroup() + 37
24  libCoin.80.dylib              	0x000000010740eb4e SoSeparator::~SoSeparator() + 14
25  libCoin.80.dylib              	0x000000010739580c SoBase::destroy() + 172
26  libCoin.80.dylib              	0x0000000107388f1a SoBaseList::truncate(int) + 74
27  libCoin.80.dylib              	0x0000000107399d01 SoChildList::~SoChildList() + 17
28  libCoin.80.dylib              	0x00000001073f0ec5 SoGroup::~SoGroup() + 37
29  libCoin.80.dylib              	0x000000010740eb4e SoSeparator::~SoSeparator() + 14
30  libCoin.80.dylib              	0x000000010739580c SoBase::destroy() + 172
31  libCoin.80.dylib              	0x0000000107388f1a SoBaseList::truncate(int) + 74
32  libCoin.80.dylib              	0x0000000107399d01 SoChildList::~SoChildList() + 17
33  libCoin.80.dylib              	0x00000001073c90a5 SoBaseKit::~SoBaseKit() + 37
34  libFreeCADGui.dylib           	0x00000001049b1ede Gui::SoFCCSysDragger::~SoFCCSysDragger() + 14
35  libCoin.80.dylib              	0x000000010739580c SoBase::destroy() + 172
36  libFreeCADGui.dylib           	0x0000000104784117 Gui::TaskCSysDragger::~TaskCSysDragger() + 39
37  libFreeCADGui.dylib           	0x00000001047845de Gui::TaskCSysDragger::~TaskCSysDragger() + 14
38  libFreeCADGui.dylib           	0x0000000104a05f59 Gui::TaskView::TaskView::removeDialog() + 361
39  libFreeCADGui.dylib           	0x0000000104a06d9a Gui::TaskView::TaskView::accept() + 202
40  libQt5Core.5.12.9.dylib       	0x0000000106dacd64 QMetaObject::activate(QObject*, int, int, void**) + 1800

OS: macOS Mojave (10.14)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

The problem does NOT occur in master FC 0.20.

Miksu80, 2021-08-03:
Could not reproduce on 2021.07.17 daily @Win10. Can you hit with a video so we can try to reproduce exactly like you did.

thomasgi1, 2021-08-03:
I tested it again and found that the crash actually happens, when the task view panel disappears. So, in order to reproduce the issue one must set the task view panel to overlay mode with the auto hide option enabled.

Shkolik, 2021-08-05:
Tested on:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

For me it works. No crash on closing task panel. @thomasgi1 please upload video to be sure we are doing the same steps...

thomasgi1, 2021-08-05:
> For me it works. No crash on closing task panel. @thomasgi1 please upload video to be sure we are doing the same steps...

Here it is:
[video](https://drive.google.com/file/d/1YPvpxCKl_iQybamOPsppAASRoOg8tb7X/view?usp=sharing)

Note that the crash happened in the video, when I clicked somewhere in the 3D view.

Miksu80, 2021-08-06:
Thanks for the vid. I can't reproduce on windows..

MAC testers needed!

Shkolik, 2021-08-10:
Yep, works fine on windows. 

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

Will try to test on mac this evening.

domomib, 2021-08-31:
the same procedure but with an object, I provided the screen capture and the freecad file.

1. I create a link to body named pieds
2. I open transform window 
3. I turn around the Z-axis to 180°
4. I click OK to validate the transformation
5. Th application crash with the following log

Process:               FreeCAD [1414]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/FreeCAD
Identifier:            FreeCAD
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           FreeCAD [1414]
User ID:               502

Date/Time:             2021-08-31 20:26:15.956 +0200
OS Version:            Mac OS X 10.13.6 (17G66)
Report Version:        12
Anonymous UUID:        C75BE5EC-87BE-4110-FE4C-FE4668B21493

Sleep/Wake UUID:       ADDBFDFD-948F-4983-B93D-E0645C23BD7E

Time Awake Since Boot: 15000 seconds
Time Since Wake:       960 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x10:
--> 
    __TEXT                 000000010276c000-0000000102773000 [   28K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/FreeCAD

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   libCoin.80.dylib              	0x0000000105395e50 SoCallbackAction::~SoCallbackAction() + 144
1   libCoin.80.dylib              	0x0000000105395fbe SoCallbackAction::~SoCallbackAction() + 14
2   libCoin.80.dylib              	0x000000010540fd12 SoDragger::~SoDragger() + 194
3   libFreeCADGui.dylib           	0x0000000102b8db0e Gui::RDragger::~RDragger() + 14
4   libCoin.80.dylib              	0x000000010557280c SoBase::destroy() + 172
5   libCoin.80.dylib              	0x0000000105565f1a SoBaseList::truncate(int) + 74
6   libCoin.80.dylib              	0x0000000105576d01 SoChildList::~SoChildList() + 17
7   libCoin.80.dylib              	0x00000001055cdec5 SoGroup::~SoGroup() + 37
8   libCoin.80.dylib              	0x00000001055ebb4e SoSeparator::~SoSeparator() + 14
9   libCoin.80.dylib              	0x000000010557280c SoBase::destroy() + 172
10  libCoin.80.dylib              	0x0000000105565f1a SoBaseList::truncate(int) + 74
11  libCoin.80.dylib              	0x0000000105576d01 SoChildList::~SoChildList() + 17
12  libCoin.80.dylib              	0x00000001055cdec5 SoGroup::~SoGroup() + 37
13  libCoin.80.dylib              	0x00000001055efe18 SoSwitch::~SoSwitch() + 56
14  libCoin.80.dylib              	0x000000010557280c SoBase::destroy() + 172
15  libCoin.80.dylib              	0x0000000105566058 SoBaseList::remove(int) + 40
16  libCoin.80.dylib              	0x00000001055ce3d7 SoGroup::removeChild(int) + 71
17  libFreeCADGui.dylib           	0x0000000102c8e59f Gui::coinRemoveAllChildren(SoGroup*) + 735
18  libFreeCADGui.dylib           	0x0000000102b6f39b Gui::SoFCPathAnnotation::~SoFCPathAnnotation() + 27
19  libFreeCADGui.dylib           	0x0000000102b6f62e Gui::SoFCPathAnnotation::~SoFCPathAnnotation() + 14
20  libCoin.80.dylib              	0x000000010557280c SoBase::destroy() + 172
21  libCoin.80.dylib              	0x0000000105565f1a SoBaseList::truncate(int) + 74
22  libCoin.80.dylib              	0x0000000105576d01 SoChildList::~SoChildList() + 17
23  libCoin.80.dylib              	0x00000001055cdec5 SoGroup::~SoGroup() + 37
24  libCoin.80.dylib              	0x00000001055ebb4e SoSeparator::~SoSeparator() + 14
25  libCoin.80.dylib              	0x000000010557280c SoBase::destroy() + 172
[... 815 lines cut ...]
MALLOC_LARGE (reserved)            384K        2         reserved VM address space (unallocated)
Memory Tag 242                      12K        2 
OpenGL GLSL                        256K        4 
PROTECTED_MEMORY                     4K        2 
STACK GUARD                       56.1M       25 
Stack                            154.7M       25 
VM_ALLOCATE                      454.1M      106 
VM_ALLOCATE (reserved)           768.0M        4         reserved VM address space (unallocated)
__DATA                            68.1M      551 
__FONT_DATA                          4K        2 
__GLSLBUILTINS                    2588K        2 
__LINKEDIT                       243.2M      216 
__TEXT                           532.0M      496 
__UNICODE                          560K        2 
mapped file                       75.4M       29 
shared memory                     2804K       18 
===========                     =======  ======= 
TOTAL                              2.7G     1603 
TOTAL, minus reserved VM space     2.0G     1603 

Model: iMac11,3, BootROM 99.0.0.0.0, 4 processors, Intel Core i7, 2,93 GHz, 16 GB, SMC 1.59f2

[relais.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7086024/relais.FCStd.zip)
 
[image]

domomib, 2021-09-13:
same issue on version daily 2021.09.11

closed 2021-09-15 by realthunder