realthunder/FreeCAD_assembly3 #766: Freecad ifc4 to ifc2x3

marcin-1981, 2021-08-02:
Hello
I'm having trouble converting from setp to ifc.
The current version of Freecad uses ifc4 schema, but my BOCAD program does not accept this version. Is it possible to change to ifc2x3 schema? Or is there a freecad option that supports this older version?

realthunder, 2021-08-02:
I am not familiar with IFC format. @yorikvanhavre Hi yorik, any idea about this?

yorikvanhavre, 2021-08-02:
Oops, it used to be configurable in the preferences but I see the entry has been removed... I'll add it again.
In the meantime you can do that manually: Go to Tools -> Preferences editor, and under BaseApp -> Preferences -> Mod -> Arch, add a new Integer value named "ifcVersion" and set it to 0 for IFC4 or 1 for IFC2x3. After that when exporting IFC files it should be what yo configured there.
[image]

marcin-1981, 2021-08-02:
Great, thank you it works.

marcin-1981, 2021-08-02:
@yorikvanhavre  Everything works under Windows 10, but I want to do it on another computer with Windows 7 and this option doesn't work. Do you know a way for this to work under Windows 7?

yorikvanhavre, 2021-08-02:
Maybe that version of FreeCAD is too old?

yorikvanhavre, 2021-08-02:
I do not know which FreeCAD version this corresponds too, so it's hard for me to help...

marcin-1981, 2021-08-02:
@yorikvanhavre then what version of freecad do this do to run under windows 7?

marcin-1981, 2021-08-02:
@yorikvanhavre I also installed freecad 0.19.2 from the manufacturer's website and also after changing this function it does not work on windows 7.

marcin-1981, 2021-08-02:
@yorikvanhavre  I guess, I'm doning this right?
[image]

[image]

marcin-1981, 2021-08-03:
@yorikvanhavre Can You help me with this?

luzpaz, 2021-08-03:
@marcin-1981  Did you make the change, save and then restart FreeCAD?

Also try installing the latest development version from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

marcin-1981, 2021-08-03:
@luzpaz Yes of course.

luzpaz, 2021-08-03:
@marcin-1981 please confirm on the development version (not LinkStage3)

marcin-1981, 2021-08-03:
On the development version, the change also has no effect.

luzpaz, 2021-08-03:
@yorikvanhavre shall I open a ticket on mantis?

yorikvanhavre, 2021-08-03:
I don't understand it working on Windows10 but not on Windows7... Can you try using the exact same FreeCAD version on both machines?

marcin-1981, 2021-08-03:
@yorikvanhavre I do it on the same software version.  Is there any other way to enable this IFC format change?

yorikvanhavre, 2021-08-03:
There isn't any other way, we would need to debug what's going wrong between the two versions... Can you ty this on both machines, in the Python console of FreeCAD?
```
import ifcopenshell
ifcopenshell.version
```
This should print a version number, ex. `'0.6.0b0'`

marcin-1981, 2021-08-03:
windows 7:
[image]

windows10:
[image]

marcin-1981, 2021-08-04:
@yorikvanhavre Will it be possible to somehow solve the situation by joint forces? ;)

yorikvanhavre, 2021-08-04:
Honestly I fail to see the cause of the problem... Could you try again with the exact same file on both machines doing exactly the same process, and verify that the two files are different? I cannot think of a reason why the exact same code would produce two different results on a same OS...

luzpaz, 2021-08-06:
@marcin-1981 :point_up_2:

marcin-1981, 2021-08-06:
@luzpaz yes? I have no changes.

luzpaz, 2021-08-06:
@marcin-1981 I pinged you because you didn't respond to yorik's request to 'try again'

marcin-1981, 2021-08-06:
@luzpaz  I've done what he asked me. And I have no changes.

luzpaz, 2021-08-06:
@marcin-1981 Do you know any other window7 users who can test this?

marcin-1981, 2021-08-06:
@luzpaz  I tested it on two computers with WINDOWS 7 with the same result. Other computers have windows 10 and it works for them.

luzpaz, 2021-08-06:
@yorikvanhavre lol

marcin-1981, 2021-08-06:
@luzpaz it would be great if someone could check under Windows 7 if they also have this problem.