realthunder/FreeCAD_assembly3 #767: A little bug with additive box top view at starting FreeCAD

mwlandolsi, 2021-08-02:
when the Freecad is set to start with a new opened file, creating an additive element is not well displayed "as is" it just shows edges not faces.

https://user-images.githubusercontent.com/76944804/127920788-30bf4228-84dd-4773-a30c-b1a0a7ad3805.mp4

bf5man, 2021-08-11:
I could not reproduce it on my side on W10.

OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: French/Canada (fr_CA)

kevenwyld, 2021-08-11:
So this is a weird one.  I was able to reproduce it but only with very specific configuration.

It's only an issue with:
* experimental rendering turned on
* when orthographic rendering is enabled as the default (not perspective) in the preferences dialogue
* when "create new document at start up" is checked in general->document->general
* NOTE: If you change the view from ortho to perspective with the cube menu the issue goes away for the current document but comes back when you create a new one.

I usually use perspective so I did not notice it at first.  I think the document on startup is important because it sets the view as "top" from the start, where as a regular document is created with the view in isometric.  

Here's a video:

https://user-images.githubusercontent.com/11936535/129107827-f7b76503-a3c0-43bd-b65e-05552db13995.mp4

realthunder, 2021-08-11:
There is a recent fix of bounding box calculation, which is probably the cause of this problem. Can you see the full object when you hit 'Fit view'?

kevenwyld, 2021-08-12:
@realthunder yes clicking the fit view button seems to resolve it for the current document.

https://user-images.githubusercontent.com/11936535/129120127-8f54a411-7f54-4189-80e3-b5f9e384c607.mp4

closed 2021-09-28 by mwlandolsi