realthunder/FreeCAD_assembly3 #769: FreeCAD crashes on LinuxMint during normal ComboView usage

Geitde, 2021-08-03:
At least that is what I imaging, as it just crashes without performing any specific operation or rendering. 

As you requested on YT toay here is the shell log.

I loaded the PartDesign example direct from start workbench and added a circle to the first sketches "tower". A simple test I always do to check if the TNP is enabled in FreeCAD. This worked. I rotated a little. No problem. I opened the pad/pockets in ComboView one by one and boom.

This is reproducible with any model and any recent version from this repository. Loading a model is not required, just the easiest way to reproduce. 

It worked before and your branch is a stunning experience to work with and a glimpse what FreeCAD could become. Keep up the great work and we hopefully get your TN solution in official FreeCAD soon. 

FreeCAD 0.19, Libs: 2021.717R24301 +3627 (Git)
jim@LINUXMINT:~$ '/home/jim/Schreibtisch/FreeCAD-asm3-Stable-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage' 
/tmp/.mount_FreeCA02y7wL
FreeCAD 0.19, Libs: 2021.717R24301 +3627 (Git)
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Body
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Sketch
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Pad
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Pad
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Sketch001
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Pocket
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Pocket
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Sketch002
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Pocket001
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Pocket001
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Sketch003
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Pocket002
<PropShape> PropertyTopoShape.cpp(398): Pending recompute for generating element map: PartDesignExample#Pocket002
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f2007189210]
#1  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x1eb692) [0x7f1ffc8b2692]
#2  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x1ebcf3) [0x7f1ffc8b2cf3]
#3  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x10cc59) [0x7f1ffc7d3c59]
#4  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x1dac7c) [0x7f1ffc8a1c7c]
#5  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x1dc9a6) [0x7f1ffc8a39a6]
#6  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x1d133c) [0x7f1ffc89833c]
#7  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x11f08c) [0x7f1ffc7e608c]
#8  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x1d0734) [0x7f1ffc897734]
#9  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x2dcd49) [0x7f1ffc9a3d49]
#10  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x2dcff3) [0x7f1ffc9a3ff3]
#11  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x3fdf40) [0x7f1ffcac4f40]
#12  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x3d4495) [0x7f1ffca9b495]
#13  /usr/lib/x86_64-linux-gnu/dri/i915_dri.so(+0x35a5e7) [0x7f1ffca215e7]
#14  0x7f2006c97511 in SoGLModelMatrixElement::translateEltBy(SbVec3f const&) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x21
#15  0x7f2006bfe038 in SoModelMatrixElement::translateBy(SoState*, SoNode*, SbVec3f const&) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0xe8
#16  0x7f2006d48245 in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x3e5
#17  0x7f2006d48245 in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x3e5
#18  0x7f200963247b in Gui::SoFCPathAnnotation::GLRenderBelowPath(SoGLRenderAction*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x3eb
#19  0x7f2006d2c4bb in SoGroup::GLRender(SoGLRenderAction*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x1ab
#20  0x7f2009616066 in Gui::SoFCSelection::glRender(SoGLRenderAction*, bool) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0xd06
#21  0x7f2006d48245 in SoSeparator::GLRenderBelowPath(SoGLRenderAction*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x3e5
#22  /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so(+0x91ee59) [0x7f2009784e59]
#23  0x7f2006b1d149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#24  0x7f2006cdea3a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x16a
#25  0x7f20096332ef in Gui::SoFCSwitch::doAction(SoAction*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x7ff
#26  /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so(+0x7cb2d0) [0x7f20096312d0]
#27  0x7f2009631bfe in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x8be
#28  0x7f2009630afd in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0xcd
#29  0x7f2006d2c3da in SoGroup::GLRender(SoGLRenderAction*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0xca
#30  0x7f2006b1d149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#31  0x7f2006cde952 in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x82
#32  0x7f2009632fc6 in Gui::SoFCSwitch::doAction(SoAction*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x4d6
#33  /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so(+0x7cb2d0) [0x7f20096312d0]
#34  0x7f2009631bfe in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x8be
#35  0x7f2009630afd in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0xcd
#36  /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so(+0x7cb2d0) [0x7f20096312d0]
#37  0x7f2009631bfe in Gui::SoFCSelectionRoot::_renderPrivate(SoGLRenderAction*, bool, bool&) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x8be
#38  0x7f2009630afd in Gui::SoFCSelectionRoot::renderPrivate(SoGLRenderAction*, bool) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0xcd
#39  0x7f2006b1d149 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x69
#40  0x7f2006cde952 in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/../lib/libCoin.so.80+0x82
#41  0x7f2009632fc6 in Gui::SoFCSwitch::doAction(SoAction*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x4d6
[... 44 lines cut ...]
#86  /tmp/.mount_FreeCA02y7wL/usr/bin/../lib/libQt5Widgets.so.5(+0x17efaf) [0x7f2007eb3faf]
#87  0x7f2007ec4dd4 in QWidgetPrivate::syncBackingStore() from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib/libQt5Widgets.so.5+0x66
#88  0x7f2007ed4cd9 in QWidget::event(QEvent*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib/libQt5Widgets.so.5+0x91f
#89  0x7f2007f7f34f in QMainWindow::event(QEvent*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib/libQt5Widgets.so.5+0xdd
#90  0x7f20097d8f9c in Gui::MainWindow::event(QEvent*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x7c
#91  0x7f2007ea97f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#92  0x7f2007eaf31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#93  0x7f200936e569 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x89
[... cut ...]
#102  0x7f2007681e8f in QCoreApplication::exec() from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib/libQt5Core.so.5+0xfd
#103  0x7f20092e7e1e in Gui::Application::runApplication() from /tmp/.mount_FreeCA02y7wL/usr/bin/../lib64/libFreeCADGui.so+0x40e
#104  /tmp/.mount_FreeCA02y7wL/usr/bin/FreeCADLink(+0x44bf) [0x5564fb33b4bf]
#105  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f200716a0b3]
#106  /tmp/.mount_FreeCA02y7wL/usr/bin/FreeCADLink(+0x47d9) [0x5564fb33b7d9]
jim@LINUXMINT:~$ ^C
jim@LINUXMINT:~$ ^C
jim@LINUXMINT:~$ ^C
jim@LINUXMINT:~$

realthunder, 2021-08-03:
Thanks for the report. But I can't reproduce the problem right away. Can you please send me you `link.user.cfg` file in your `~/.FreeCAD` directory. Thanks.

Geitde, 2021-08-03:
Here is the file. I hope it works this way.

[link.user.cfg.zip](https://github.com/realthunder/FreeCAD_assembly3/files/6924797/link.user.cfg.zip)