realthunder/FreeCAD_assembly3 #771: Color rendering issue

kenjesse, 2021-08-05:
Having an odd issue with how bodies and other items color are being rendered on my laptop.  Program functionality is not affected beyond visual problem.  I have attached two images showing what is happening (colored wire frame around body or lines rather than applying the color).  I have tried resetting the render cache to all available setting, restarting the program after each change, with no impact.
[image]
[image]

Lenovo Ideapad, AMD A9-9400 Radeon R5 (2-cores), 8GB RAM
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

thomasgi1, 2021-08-05:
It's not a bug. It's a feature. Just disable the bounding box for the selection:

View / Selection Options / Bounding Box => disable

BR Tom

kenjesse, 2021-08-06:
Very new to not just Freecad but CAD in general.  Had no success searching for info and posted here.  Thanks for the help, kind of embarrassed I missed it while searching through the setting.

closed 2021-08-06 by kenjesse