realthunder/FreeCAD_assembly3 #772: included Example projects trigger errors

dmerillat, 2021-08-05:
Tested with Daily-20210717.

EngineBlock:
```
18:53:54  Traceback (most recent call last):
  File "/tmp/.mount_FreeCA1XGoq2/usr/Mod/Draft/draftobjects/bspline.py", line 109, in execute
    spline.interpolate(obj.Points, PeriodicFlag = True, Parameters = self.knotSeq)
<class 'Part.OCCError'>: Standard_ConstructionError
18:53:54  Recompute failed! Please check report view.
```

ArchDetail:
```
19:01:08  <Exception> Sketch::checkGeoId. GeoId index out range.
19:01:08  Traceback (most recent call last):
  File "/tmp/.mount_FreeCA1XGoq2/usr/Mod/Draft/draftobjects/array.py", line 406, in execute
    obj.IntervalZ,
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'IntervalZ'
19:01:08  Recompute failed! Please check report view.
```

RobotExample1: 
recalculate failed with no other information given

I know daily is going through a lot of changes so this isn't very high priority, just to doublecheck when the big work gets moved to stable.

luzpaz, 2021-08-06:
Please add the file

Shkolik, 2021-08-10:
> Please add the file

As I understand it's example files shipped with freecad

luzpaz, 2021-08-10:
Can you recreate with the latest [weekly build](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) for the master branch?

dmerillat, 2021-08-10:
> > Please add the file
> 
> @luzpaz, As I understand it's example files shipped with freecad

Correct, it is the files that show up on the start screen as examples.

> Can you recreate with the latest [weekly build](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) for the master branch?

No, the latest weekly (25429, March 3 2021) does not have the same errors when loading the example projects.

Reproduction is two steps:
1) open freecad
2) double click example project

luzpaz, 2022-03-19:
Some of the examples in the start wb are problematic.