realthunder/FreeCAD_assembly3 #773: FreeCAD example FemCalculixCantilever3D crash on recompute

nekopep, 2021-08-06:
Hi there,
Just seen that when I open FemCalculixCantilever3D from FreeCad start workbench, it ask a recompute and then crash immediately.
Here is the traceback:
./FreeCAD.app/Contents/MacOS/FreeCADLink --system-cfg $PWD/link.system.cfg --user-cfg $PWD/link.user.cfg
FreeCAD 0.19, Libs: 2021.717R24301 +3842 (Git)
<PropShape> PropertyTopoShape.cpp(412): Pending recompute for generating element map: FemCalculixCantilever3D#Box
./FreeCAD.app/Contents/MacOS/FreeCADLink: line 27: 88638 Illegal instruction: 4  "$PREFIX/bin/$name" "$@"

Since it seems related to topology I post the bug report here.

Best

luzpaz, 2021-08-06:
Can you recreate it on freeCAD master weekly build? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

nekopep, 2021-08-08:
Hello, sorry for the delay, I confirm that on freecad weekly build it works as expected.
Still i did not used my user profile.

I think this bug has occured multiple time to me while using freecad linkstage 3. I had several crash with such log, the FemCalculixCantilevever3D is just an esay way to reproduce it.

nekopep, 2021-08-08:
Ok I just restedted with same system and user profile and confirm the bug only affects linkstage3 on macosx. 
I can't test on linux now, but in 2 weeks if needed.

luzpaz, 2021-08-08:
yes please

thomasgi1, 2021-08-08:
It does NOT crash here:

OS: macOS Mojave (10.14)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

Illegal instruction sounds like an issue with compiler optimisations for specific CPU features, which your CPU doesn't have. My MacBook Air has following CPU features:

sysctl -a | grep cpu.features
machdep.cpu.features: FPU VME DE PSE TSC MSR PAE MCE CX8 APIC SEP MTRR PGE MCA CMOV PAT PSE36 CLFSH DS ACPI MMX FXSR SSE SSE2 SS HTT TM PBE SSE3 PCLMULQDQ DTES64 MON DSCPL VMX SMX EST TM2 SSSE3 FMA CX16 TPR PDCM SSE4.1 SSE4.2 x2APIC MOVBE POPCNT AES PCID XSAVE OSXSAVE SEGLIM64 TSCTMR AVX1.0 RDRAND F16C

I guess that your CPU lacks some of the SSE features.

nekopep, 2021-08-09:
Here my infos:

I'm on catalina and indeed this is an old mac book pro retina from 2013 (core i7)

sysctl -a | grep cpu.features
machdep.cpu.features: FPU VME DE PSE TSC MSR PAE MCE CX8 APIC SEP MTRR PGE MCA CMOV PAT PSE36 CLFSH DS ACPI MMX FXSR SSE SSE2 SS HTT TM PBE SSE3 PCLMULQDQ DTES64 MON DSCPL VMX EST TM2 SSSE3 CX16 TPR PDCM SSE4.1 SSE4.2 x2APIC POPCNT AES PCID XSAVE OSXSAVE TSCTMR AVX1.0 RDRAND F16C

This is what is missing from my CPU:  FMA SMX MOVBE SEGLIM64

nekopep, 2021-08-09:
My freecad info:
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

thomasgi1, 2021-08-09:
I checked my crash reports today and found one related to the FemCalculixCantilever3D experiment. So, I also had a crash on my machine, but it happened, when closing the FC application. That's why I did not recognise the crash. However, here is the full stack trace for the crash:

Process:               FreeCAD [37328]
Path:                  /Applications/FreeCAD asm3.app/Contents/Resources/bin/FreeCAD
Identifier:            ???
Version:               ??? (2021)
Code Type:             X86-64 (Native)
Parent Process:        ??? [37321]
Responsible:           FreeCAD [37328]
User ID:               501

Date/Time:             2021-08-08 23:48:46.972 +0200
OS Version:            Mac OS X 10.14.6 (18G9216)
Report Version:        12
Anonymous UUID:        00D0AAE4-4FA2-1AA2-1DC3-429829F4143B

Sleep/Wake UUID:       D6C6A42F-D2CA-4A36-B1CC-AAD68A887DD6

Time Awake Since Boot: 1300000 seconds
Time Since Wake:       16000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [37328]

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   libSMDS.dylib                 	0x000000017ab5db7b SMDS_ElementChunk::~SMDS_ElementChunk() + 283
1   libSMDS.dylib                 	0x000000017ab5bc49 SMDS_ElementFactory::Clear() + 121
2   libSMDS.dylib                 	0x000000017ab66d9a SMDS_Mesh::Clear() + 42
3   libSMESHDS.dylib              	0x0000000175dfd3a6 SMESHDS_Mesh::ClearMesh() + 54
4   libSMESH.dylib                	0x000000017b7e9c97 SMESH_Mesh::~SMESH_Mesh() + 263
5   libSMESH.dylib                	0x000000017b7ea19e SMESH_Mesh::~SMESH_Mesh() + 14
6   Fem.so                        	0x000000017acdb783 Fem::FemMesh::~FemMesh() + 131
7   Fem.so                        	0x000000017acdb9a5 non-virtual thunk to Fem::FemMesh::~FemMesh() + 21
8   Fem.so                        	0x000000017ad1078c Fem::PropertyFemMesh::~PropertyFemMesh() + 44
9   Fem.so                        	0x000000017acd3d1b App::FeaturePythonT<Fem::FemMeshObject>::~FeaturePythonT() + 107
10  libFreeCADApp.dylib           	0x000000010fe12ad5 App::Document::~Document() + 101
11  libFreeCADApp.dylib           	0x000000010fe13a2e App::Document::~Document() + 14
12  libFreeCADApp.dylib           	0x00000001100d5726 App::Application::closeDocument(char const*) + 630
13  libFreeCADApp.dylib           	0x00000001100d580f App::Application::closeAllDocuments() + 47
14  libFreeCADGui.dylib           	0x000000010ed9c074 Gui::MainWindow::closeEvent(QCloseEvent*) + 52
15  libQt5Widgets.5.12.9.dylib    	0x00000001106628fa QWidget::event(QEvent*) + 2138
16  libFreeCADGui.dylib           	0x000000010ed9a15b Gui::MainWindow::event(QEvent*) + 1035

Thread 1:: com.apple.CFSocket.private
0   libsystem_kernel.dylib        	0x00007fff70aab616 __select + 10
1   com.apple.CoreFoundation      	0x00007fff449e588a __CFSocketManager + 635
2   libsystem_pthread.dylib       	0x00007fff70b632eb _pthread_body + 126
3   libsystem_pthread.dylib       	0x00007fff70b66249 _pthread_start + 66
4   libsystem_pthread.dylib       	0x00007fff70b6240d thread_start + 13

Thread 2:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff70aa421a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff70aa4768 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff449b7e0e __CFRunLoopServiceMachPort + 328
3   com.apple.CoreFoundation      	0x00007fff449b737c __CFRunLoopRun + 1612
4   com.apple.CoreFoundation      	0x00007fff449b6ade CFRunLoopRunSpecific + 455
5   com.apple.AppKit              	0x00007fff41fb44a2 _NSEventThread + 175
6   libsystem_pthread.dylib       	0x00007fff70b632eb _pthread_body + 126
7   libsystem_pthread.dylib       	0x00007fff70b66249 _pthread_start + 66
8   libsystem_pthread.dylib       	0x00007fff70b6240d thread_start + 13

[... 906 lines cut ...]
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       286.8M      302 
__TEXT                           605.0M      607 
__UNICODE                          564K        1 
mapped file                      457.1M       99 
shared memory                      892K       18 
===========                     =======  ======= 
TOTAL                              2.5G     2241 
TOTAL, minus reserved VM space     2.3G     2241 

System Profile:
Network Service: Wi-Fi, AirPort, en0
Thunderbolt Bus: MacBook Air, Apple Inc., 27.2
Boot Volume File System Type: apfs
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E424A544D4C41522D4E5544
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E424A544D4C41522D4E5544
USB Device: USB 3.0 Bus
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Serial ATA Device: APPLE SSD SM0512G, 500.28 GB
Model: MacBookAir7,2, BootROM 427.0.0.0.0, 2 processors, Intel Core i7, 2.2 GHz, 8 GB, SMC 2.27f2
Graphics: kHW_IntelHDGraphics6000Item, Intel HD Graphics 6000, spdisplays_builtin
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.77.61.3 AirPortDriverBrcmNIC-1305.12)
Bluetooth: Version 6.0.14d12, 3 services, 27 devices, 1 incoming serial ports

nekopep, 2021-08-10:
@thomasgi1 , FYI I got the exact same crash -> SMDS_ElementChunk::~SMDS_ElementChunk() destructor.

Crashed Thread:        0  CrBrowserMain

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [6433]

Thread 0 Crashed:: CrBrowserMain
0   libSMDS.dylib                 	0x000000018d13bb7b SMDS_ElementChunk::~SMDS_ElementChunk() + 283
1   libSMDS.dylib                 	0x000000018d139c49 SMDS_ElementFactory::Clear() + 121
2   libSMDS.dylib                 	0x000000018d144d9a SMDS_Mesh::Clear() + 42
3   libSMESHDS.dylib              	0x0000000175ac03a6 SMESHDS_Mesh::ClearMesh() + 54
4   libSMESH.dylib                	0x000000018cfd2c97 SMESH_Mesh::~SMESH_Mesh() + 263
5   libSMESH.dylib                	0x000000018cfd319e SMESH_Mesh::~SMESH_Mesh() + 14
6   Fem.so                        	0x0000000175671783 Fem::FemMesh::~FemMesh() + 131
7   Fem.so                        	0x00000001756719a5 non-virtual thunk to Fem::FemMesh::~FemMesh() + 21
8   Fem.so                        	0x00000001756a67cc Fem::PropertyFemMesh::~PropertyFemMesh() + 44
9   libFreeCADApp.dylib           	0x0000000101d7a326 App::Property::hasSetValue() + 422
10  Fem.so                        	0x000000017566d8ef Fem::FemMeshShapeNetgenObject::execute() + 735
11  libFreeCADApp.dylib           	0x0000000101c0fe44 App::DocumentObject::recompute() + 84
12  libFreeCADApp.dylib           	0x0000000101b7a382 App::Document::_recomputeFeature(App::DocumentObject*) + 130
13  libFreeCADApp.dylib           	0x0000000101b788c2 App::Document::recompute(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*> > const&, bool, bool*, int) + 3170
14  libFreeCADGui.dylib           	0x000000010055f725 boost::detail::function::void_function_obj_invoker0<Gui::Application::Application(bool)::$_0, void>::invoke(boost::detail::function::function_buffer&) + 917
15  libFreeCADApp.dylib           	0x0000000101d1f569 boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type>, std::__1::__list_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (), boost::function<void ()> >, boost::signals2::mutex> >, void*>, boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (), boost::function<void ()> >, boost::signals2::mutex> >::dereference() const + 73
16  libFreeCADApp.dylib           	0x0000000101d1f2b1 boost::signals2::detail::signal_impl<void (), boost::signals2::optional_last_value<void>, int, std::__1::less<int>, boost::function<void ()>, boost::function<void (boost::signals2::connection const&)>, boost::signals2::mutex>::operator()() + 625
17  libFreeCADApp.dylib           	0x0000000101e26110 App::Application::openDocuments(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >*, bool) + 7840
18  libFreeCADApp.dylib           	0x0000000101e241bb App::Application::openDocument(char const*, bool) + 235
19  libFreeCADApp.dylib           	0x0000000101e559af App::Application::sOpenDocument(_object*, _object*, _object*) + 271
20  libpython3.9.dylib            	0x00000001017f0534 cfunction_call + 52
21  libpython3.9.dylib            	0x000000010178f318 _PyObject_MakeTpCall + 136
22  libpython3.9.dylib            	0x00000001018cc003 call_function + 323
23  libpython3.9.dylib            	0x00000001018c9a5c _PyEval_EvalFrameDefault + 44668
24  libpython3.9.dylib            	0x00000001018bd394 _PyEval_EvalCode + 548
25  libpython3.9.dylib            	0x00000001019325e1 PyRun_StringFlags + 337
26  libFreeCADBase.dylib          	0x00000001021d4c6b Base::InterpreterSingleton::runString(char const*) + 91
27  libFreeCADApp.dylib           	0x0000000101e554c0 App::Application::sLoadFile(_object*, _object*) + 2112
28  libpython3.9.dylib            	0x00000001017f055a cfunction_call + 90
29  libpython3.9.dylib            	0x000000010178f318 _PyObject_MakeTpCall + 136
30  libpython3.9.dylib            	0x00000001018cc003 call_function + 323
31  libpython3.9.dylib            	0x00000001018c9a5c _PyEval_EvalFrameDefault + 44668
32  libpython3.9.dylib            	0x00000001018bd394 _PyEval_EvalCode + 548
33  libpython3.9.dylib            	0x00000001019325e1 PyRun_StringFlags + 337
[... cut ...]
40  libFreeCADBase.dylib          	0x00000001021d4c6b Base::InterpreterSingleton::runString(char const*) + 91
41  libFreeCADGui.dylib           	0x0000000100666c12 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 162
42  libFreeCADGui.dylib           	0x0000000100666a4d Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) + 221
43  WebGui.so                     	0x000000016b83b635 WebGui::BrowserView::urlFilter(QUrl const&) + 1765
44  libQt5Core.5.12.9.dylib       	0x0000000102d3785d QObject::event(QEvent*) + 687

nekopep, 2021-08-10:
Ok seems related to:
https://github.com/trelau/SMESH/issues/37
https://github.com/realthunder/FreeCAD_assembly3/issues/698

@luzpaz : can you mark as duplicate of 698?

nekopep, 2021-08-10:
Also can you remove "waiting user response"? thx!

luzpaz, 2022-01-19:
Closing as duplicate of #698

closed 2022-01-19 by luzpaz