realthunder/FreeCAD_assembly3 #776: Program received signal SIGSEGV, Segmentation fault.

snelweg, 2021-08-09:
I compiled your Linkstage3 branch myself hoping to get rid of the error and salvage the file, but I am sorry to say the file still won't open. I assumed the fix is included in the latest git clone and the file could be opened with the latest version straight from git.

> OS: Debian GNU/Linux 10 (buster) (X-Cinnamon/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27928 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

> ton@DESKTOP-9FIBCI1:~/FREECAD/freecad-build/bin$ ./FreeCAD
> FreeCAD 0.19, Libs: 0.19R27928 (Git)
> © Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
> FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
> FreeCAD wouldn't be possible without FreeCAD community.
> Sheet Metal workbench loaded
> <PropShape> PropertyTopoShape.cpp(382): Recomputation required for document 'verwarming' on geo element version change in verwarming#Body.Shape: 1.15.70200.4 -> 1.15.70200.3.0
> Program received signal SIGSEGV, Segmentation fault.
> #0  /lib/x86_64-linux-gnu/libc.so.6(+0x37840) [0x7f536d6e3840]
> #1  0x7f5371853644 in App::PropertyLink::getValue() const from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xc
> #2  0x7f53717cbd69 in App::LinkBaseExtension::syncElementList() from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x321
> #3  0x7f53717ca79d in App::LinkBaseExtension::update(App::DocumentObject*, App::Property const*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x19b5
> #4  0x7f53717cc7f6 in App::LinkBaseExtension::onExtendedDocumentRestored() from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x9a2
> #5  0x7f537165c128 in void App::ExtensionContainer::callExtension<void (App::DocumentObjectExtension::*)()>(void (App::DocumentObjectExtension::*&&)()) const from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xf8
> #6  0x7f5371657a9d in App::DocumentObject::onDocumentRestored() from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x31
> #7  0x7f537178de5c in App::GeoFeature::onDocumentRestored() from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xa4
> #8  0x7f5367a86e67 in App::FeaturePythonT<Part::Feature>::onDocumentRestored() from /home/ton/FREECAD/freecad-build/Mod/Part/Part.so+0x2b
> #9  0x7f5371557e8d in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x60f
> #10  0x7f537155769b in App::Document::afterRestore(bool) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0x5d
> #11  0x7f53718edb3f in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xaf7
> #12  0x7f53718ec789 in App::Application::openDocument(char const*, bool) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xc5
> #13  0x7f5371926ebd in App::Application::sOpenDocument(_object*, _object*, _object*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADApp.so+0xe9
> #14  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyMethodDef_RawFastCallKeywords+0x30b) [0x7f536fdc3cdb]
> #15  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyCFunction_FastCallKeywords+0x25) [0x7f536fdc4a05]
> #16  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalFrameDefault+0x7e90) [0x7f536fbc5cb0]
> #17  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(_PyEval_EvalCodeWithName+0x996) [0x7f536fced1e6]
> #18  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(PyEval_EvalCodeEx+0x3e) [0x7f536fced46e]
> #19  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(PyEval_EvalCode+0x1b) [0x7f536fcee23b]
> #20  /lib/x86_64-linux-gnu/libpython3.7m.so.1.0(PyRun_StringFlags+0x8b) [0x7f536fcbbfeb]
> #21  0x7f537081b302 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADBase.so+0xd8
> #22  0x7f5372f7fcc6 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0xca
> #23  0x7f5372f7f990 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x112
> #24  0x7f5372dcc1a1 in Gui::Application::open(char const*, char const*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x3df
> #25  0x7f5372f51b2b in Gui::RecentFilesAction::activateFile(int) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x28f
> #26  0x7f5372fb15c7 in StdCmdRecentFiles::activated(int) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x3b
> #27  0x7f5372f7e8f8 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x3be
> #28  0x7f5372f4ed45 in Gui::ActionGroup::onActivated(QAction*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x69
> #29  /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so(+0x1258769) [0x7f5372f58769]
> #30  0x7f536de8e7f6 in QMetaObject::activate(QObject*, int, int, void**) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x766
> #31  0x7f536e7df6af in QActionGroup::triggered(QAction*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x3f
> #32  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x15d161) [0x7f536e7e0161]
> #33  0x7f536de8e7f6 in QMetaObject::activate(QObject*, int, int, void**) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x766
> #34  0x7f536e7dbf02 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
> #35  0x7f536e7de520 in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf0
> #36  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2cba4c) [0x7f536e94ea4c]
> #37  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2d2f70) [0x7f536e955f70]
> #38  0x7f536e956edb in QMenu::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x26b
> #39  0x7f536e8204d8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1d8
> #40  0x7f536e95936b in QMenu::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x11b
> #41  0x7f536e7e24c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x81
> #42  0x7f536e7e9bb8 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x458
> #43  0x7f5372f1dd3d in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0xff
> #44  0x7f536de65489 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
> #45  0x7f536e7e8ee9 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a9
> #46  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1b8393) [0x7f536e83b393]
> #47  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1bab5e) [0x7f536e83db5e]
> #48  0x7f536e7e24c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x81
> #49  0x7f536e7e9970 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x210
> #50  0x7f5372f1dd3d in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0xff
> #51  0x7f536de65489 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
[... cut ...]
> #57  0x7f536de6c132 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
> #58  0x7f5372dd6e78 in Gui::Application::runApplication() from /home/ton/FREECAD/freecad-build/lib/libFreeCADGui.so+0x5f2
> #59  ./FreeCAD(+0x9158) [0x56314fb6b158]
> #60  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f536d6d009b]
> #61  ./FreeCAD(+0x84ca) [0x56314fb6a4ca]

_Originally posted by @snelweg in https://github.com/realthunder/FreeCAD_assembly3/issues/744#issuecomment-895077879_

domomib, 2021-08-10:
I don't know if it is the same error but in my case after the crash is when : I start freecad and I close the document opened. that's all.

OS: macOS High Sierra (10.13)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

Traceback:

Process:               FreeCAD [7053]
Path:                  /Users/USER/Desktop/FreeCAD.app/Contents/Resources/bin/FreeCAD
Identifier:            ???
Version:               ??? (2021)
Code Type:             X86-64 (Native)
Parent Process:        ??? [7046]
Responsible:           FreeCAD [7053]
User ID:               501

Date/Time:             2021-08-10 10:13:07.990 +0200
OS Version:            Mac OS X 10.13.6 (17G14042)
Report Version:        12
Anonymous UUID:        8C926C31-556F-D1DC-078D-7CAAB2652FC3

Sleep/Wake UUID:       D48C9EA4-90A3-41FC-8EE0-6AD1F758754D

Time Awake Since Boot: 75000 seconds
Time Since Wake:       910 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x10:
--> 
    __TEXT                 00000001004be000-00000001004c5000 [   28K] r-x/r-x SM=COW  /Users/USER/Desktop/FreeCAD.app/Contents/Resources/bin/FreeCAD

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libCoin.80.dylib              	0x00000001030e2e10 SoCallbackAction::~SoCallbackAction() + 80
1   libCoin.80.dylib              	0x00000001030e2fbe SoCallbackAction::~SoCallbackAction() + 14
2   libFreeCADGui.dylib           	0x000000010087b24d SoFCRenderCacheManagerP::~SoFCRenderCacheManagerP() + 29
3   libFreeCADGui.dylib           	0x000000010087b501 SoFCRenderCacheManager::~SoFCRenderCacheManager() + 33
4   libFreeCADGui.dylib           	0x00000001008aad48 Gui::SoFCUnifiedSelection::~SoFCUnifiedSelection() + 40
5   libFreeCADGui.dylib           	0x00000001008aadce Gui::SoFCUnifiedSelection::~SoFCUnifiedSelection() + 14
6   libCoin.80.dylib              	0x00000001032bf80c SoBase::destroy() + 172
7   libCoin.80.dylib              	0x00000001032b2f1a SoBaseList::truncate(int) + 74
8   libCoin.80.dylib              	0x00000001032c3d01 SoChildList::~SoChildList() + 17
9   libCoin.80.dylib              	0x000000010331aec5 SoGroup::~SoGroup() + 37
10  libCoin.80.dylib              	0x0000000103338b4e SoSeparator::~SoSeparator() + 14
11  libCoin.80.dylib              	0x00000001032bf80c SoBase::destroy() + 172
12  libCoin.80.dylib              	0x00000001030eaa26 SoHandleEventAction::~SoHandleEventAction() + 54
13  libCoin.80.dylib              	0x00000001030eaa9e SoHandleEventAction::~SoHandleEventAction() + 14
14  libCoin.80.dylib              	0x00000001032dfeb3 SoEventManager::~SoEventManager() + 243
15  libCoin.80.dylib              	0x00000001032e011e SoEventManager::~SoEventManager() + 14
16  libFreeCADGui.dylib           	0x00000001009567bc SIM::Coin3D::Quarter::QuarterWidget::~QuarterWidget() + 252
[... 611 lines cut ...]
Memory Tag 242                      12K        2 
OpenGL GLSL                        256K        4 
PROTECTED_MEMORY                     4K        2 
STACK GUARD                       56.0M       10 
Stack                             27.1M       12 
VM_ALLOCATE                      449.1M      301 
VM_ALLOCATE (reserved)           768.0M        4         reserved VM address space (unallocated)
__DATA                            68.1M      551 
__FONT_DATA                          4K        2 
__GLSLBUILTINS                    2588K        2 
__LINKEDIT                       243.6M      213 
__TEXT                           531.3M      495 
__UNICODE                          560K        2 
mapped file                      663.2M      206 
shared memory                      696K       13 
===========                     =======  ======= 
TOTAL                              3.0G     1931 
TOTAL, minus reserved VM space     2.3G     1931 

Model: iMac11,3, BootROM 99.0.0.0.0, 4 processors, Intel Core i7, 2,93 GHz, 16 GB, SMC 1.59f2
Graphics: ATI Radeon HD 5750, ATI Radeon HD 5750, PCIe
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x80CE, 0x4D34373142353237334348302D4348392020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1333 MHz, 0x80CE, 0x4D34373142353237334348302D4348392020
Memory Module: BANK 0/DIMM1, 4 GB, DDR3, 1333 MHz, 0x80CE, 0x4D34373142353237334448302D4348392020
Memory Module: BANK 1/DIMM1, 4 GB, DDR3, 1333 MHz, 0x80CE, 0x4D34373142353237334448302D4348392020

snelweg, 2021-08-10:
@domomib I don  not think it is not related. There are many differences to the different architectures, for example I can open the file under Windows and save it. But when I try to open it in appimage Linux or even my own compiled version from @realthunder github it crashes. I hoped I could save it in the windows version and continue to work on my big Linux machine, but that just doesn't work.

realthunder, 2021-08-10:
> I assumed the fix is included in the latest git clone and the file could be opened with the latest version straight from git.

My development now happens in the [LinkDaily](https://github.com/realthunder/FreeCAD/tree/LinkDaily) branch. I only merge (some of the) changes back to `LinkStage3` (the default branch) before releasing. Anyway, I have just picked that fix to `LinkStage3`. Please sync and try again.

snelweg, 2021-08-10:
[update]

I made a fake file so that compiling could go on. But it didn't work. There are files missing from the path workbench.

`[ 92%] Building CXX object src/Mod/Path/Gui/CMakeFiles/PathGui.dir/ViewProviderPathShape.cpp.o
/home/ton/FREECAD/freecad-source-daily/src/Mod/Path/Gui/DlgProcessorChooser.cpp:85:10: fatal error: moc_DlgProcessorChooser.cpp: No such file or directory
 #include "moc_DlgProcessorChooser.cpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/Mod/Path/Gui/CMakeFiles/PathGui.dir/build.make:346: src/Mod/Path/Gui/CMakeFiles/PathGui.dir/DlgProcessorChooser.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 92%] Generating moc_TaskHelixParameters.cpp
/home/ton/FREECAD/freecad-source-daily/src/Mod/Path/Gui/TaskDlgPathCompound.cpp:160:10: fatal error: moc_TaskDlgPathCompound.cpp: No such file or directory
 #include "moc_TaskDlgPathCompound.cpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/Mod/Path/Gui/CMakeFiles/PathGui.dir/build.make:333: src/Mod/Path/Gui/CMakeFiles/PathGui.dir/TaskDlgPathCompound.cpp.o] Error 1
[ 92%] Generating qrc_PartDesign.cpp
[ 92%] Generating ui_TaskFeaturePick.h
In file included from /home/ton/FREECAD/freecad-source-daily/src/Mod/Path/Gui/DlgSettingsPathColor.cpp:101:
/home/ton/FREECAD/freecad-build.daily/src/Mod/Path/Gui/moc_DlgSettingsPathColor.cpp:1:1: error: expected unqualified-id before ‘-’ token
 -I/home/ton/FREECAD/freecad-build.stable
 ^
[ 96%] Built target TechDrawGui
`

snelweg, 2021-08-10:
[update2]

I removed the Path dir from the src/ and everything compiles and the file opens now! I can work on it.
Hopefully the Path workbench can be fixed, I use it a lot.

> OS: Debian GNU/Linux 10 (buster) (X-Cinnamon/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28253 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: 1fff80878668f57edf9c947ab5b9b2a26916d3ec
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

realthunder, 2021-08-10:
If you removed the `src/Mod/Path` directory in your build directory, then it will be regenerated when you build again. This is actually the recommended way of dealing with stall build in many cases.

snelweg, 2021-08-11:
> If you removed the `src/Mod/Path` directory in your build directory, then it will be regenerated when you build again. This is actually the recommended way of dealing with stall build in many cases.

Another cmake didn't regenerate the Path directory
I did a `git checkout LinkDaily Path` int src/Mod directory and this did pull in the missing Path directory

A new 
`cmake ../freecad-source-daily  -DBUILD_QT5=ON -DPYTHON_EXECUTABLE=/usr/bin/python3`
and
`cmake --build /home/ton/FREECAD/freecad-build.daily -j 6`
worked and I now have a complete and working Linkstage3 version with all your goodies and bug fixes. Happy now!

luzpaz, 2022-04-18:
bumping for progress on ticket

snelweg, 2022-04-19:
`I now have a complete and working Linkstage3 version`

closed 2022-04-19 by snelweg

luzpaz, 2022-04-19:
@snelweg do you mind notating how so that someone reading this has 'closure' ?

snelweg, 2022-04-19:
> @snelweg do you mind notating how so that someone reading this has 'closure' ?

What is missing from the post https://github.com/realthunder/FreeCAD_assembly3/issues/776#issuecomment-896605504 ???? It's all there.

>Another cmake didn't regenerate the Path directory
I did a git checkout LinkDaily Path in src/Mod directory and this did pull in the missing Path directory

A new
cmake ../freecad-source-daily -DBUILD_QT5=ON -DPYTHON_EXECUTABLE=/usr/bin/python3
and
cmake --build /home/ton/FREECAD/freecad-build.daily -j 6
worked and I now have a complete and working Linkstage3 version with all your goodies and bug fixes. Happy now!