realthunder/FreeCAD_assembly3 #778: More details on DoF and constraints in Assembly3?

slaarti, 2021-08-10:
Is there any way to get a clearer idea of where the remaining DoF in an assembly actually _are_, and/or of just how many DoF any particular constraint will remove, other than non-stop guesswork?

Just giving a number of `dof remaining` without any idea of _which ones and where_ strikes me as very unhelpful, and same for not having any idea of what the best constraints would be to resolve them without the dreaded `redundant constraints` warning popping up. (And, for that matter, if there was some indication of _which_ constraint(s) the one you just added was redundant with, that'd also be super helpful.)

henrebotha, 2022-01-10:
The Sketcher's ability to indicate which constraints are redundant is a key part of my workflow there. I'd love to see the same for A3.

ThirteenPeeps, 2022-10-17:
I can't believe there aren't more complaints about this.

I literally don't understand how anybody is able to use with workbench without flipping tables in frustration trying to read the mind of the solver about DOF. If I'm supposed to just "keep track" of which DOF exist for every element in an assembly, then I'm never using this workbench, because if I want to make an assembly, I probably already have DOF across all the elements than I can reasonably hold in my working memory.

Having the solver report how many DOF there are but not which ones, _feels like_ an arbitrary and incredibly anti-user withholding of information.