realthunder/FreeCAD_assembly3 #781: [Issue] Changing unralated sketch geometry to construction and vice versa breaks pad feature unpredictably

YosypC, 2021-08-13:
https://user-images.githubusercontent.com/50027752/129388844-9ebc73c3-2d9a-4428-8cf4-bcf9bbb21653.mp4

I tried deleting the incriminated geometry when the pad is broken, but it didn't change anything. Why does this happen? I remember having mirrored the bottom rectangle from the top one. That's it.

```

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git)
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Italian/Italy (it_IT)

```

realthunder, 2021-08-16:
Can you please post the file here?

YosypC, 2021-08-16:
Unfortunately I saved it before making a backup. I'll try replicating the issue.

YosypC, 2021-08-21:
I managed to have the same bug, this time without the changing geometry stuff unfortunately. @realthunder 
[Support.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7025321/Support.zip)