realthunder/FreeCAD_assembly3 #783: Sketcher Navigation: Tool disappears on two button mouse move

wohltat, 2021-08-17:
When moving in Sketcher with Blender Navigation style the selected tool disappears.
This is probably because the right mouse button click is interpreted as single right click that is supposed to deselect the tool. 

[image]

luzpaz, 2021-08-17:
Please always add [About](https://wiki.freecadweb.org/About). Thanks

wohltat, 2021-08-17:
```
OS: Linux Mint 20.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

```

luzpaz, 2022-03-18:
@wohltat is this still an issue?

wohltat, 2022-03-19:
The tool does not deselect anymore, but now it triggers unintentional single mouse button click while two mouse button move.
[image]

luzpaz, 2022-04-17:
@realthunder issue still persists apparently

closed 2022-04-19 by realthunder

wohltat, 2022-05-29:
Thanks, works nicely in [2022.04.30](https://github.com/realthunder/FreeCAD/releases/tag/2022.04.30-edge).