realthunder/FreeCAD_assembly3 #785: [BUG, part design] Extrusion in custom direction preview ok, resulting model not ok

Miksu80, 2021-08-18:
Do a 19mm circle on XY plane X-axis left from Y-axis so that circle rim touches sketch center. Extrude 50mm in custom direction X=0 Y=0.1 Z=1. Mirror that on XY-plane -> preview ok, resulting crash.
[image]

Resulting:
[image]

When removed "Parallel transform", it works:
[image]

File attached had a different view of the mirror -again preview OK:
[image]

Resulting in funny shape:
[image]

File for the funny shape:
[Custom_direction_mirror_fails.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7006849/Custom_direction_mirror_fails.zip)

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

realthunder, 2021-08-19:
This is most likely an OCC problem, as I can reproduce it using upstream FC. I can't reproduce the problem in your first test case. Also do you mean by mirror base on XZ plane instead of XY, because that's what's shown in your screen capture. Anyway, the difference could be due to OCC version difference. I haven't check. I am using Linux OCC 7.5.1.

For the second case, I find that OCC fails to fuse if the two cylinder bottom overlap beyond some threshold. You can check by change the transform offset X. I find that for offset smaller than -0.3mm, it works, and fails when greater than that.

BTW, the preview of pattern features shows each pattern at the desired location without further operation to speed up the process. You are supposed to turn off preview to see the end result, which may fail.

Miksu80, 2021-08-19:
Do I close this issue as it is OCC problem? or what do I do with it?