realthunder/FreeCAD_assembly3 #787: cannot set the color for the python expression window

wohltat, 2021-08-21:
If have looked in every menu and tried all possibilities i could find in the style sheets but i was not able to set the font color for the hovering python expression window. Since use a dark color for the python console background, i cannot read the expression when it is not selected anymore.
[image]

The expression window uses the same background color as the python console. I guess then it should use the same highlighting or font colors.
Furthermore i think the evaluation field on the bottom of the expression window has the wrong background color since it uses the font color(green in my case) of the "python output" / Report View but not the background color of it.

rmTizi, 2021-08-22:
Been chasing this bug for a few days, finally found the cause: 

https://github.com/realthunder/FreeCAD/blob/970eeeee9f7924ea4a3ef0a686df67e7549fa4fc/src/Gui/DlgExpressionInput.cpp#L102

Should be `color:white` instead of black.

In truth there is *a lot* of hard coded colors in that file, leaving little room to style authors to make the formula editor actually readable in alternative color schemes.

Ideally it should at least pull the colors out from the Output window and/or Text Editor user preferences and leave the background colors up the the style sheet.

realthunder, 2021-08-23:
> Been chasing this bug for a few days, finally found the cause:

@rmTizi Nice fix! Although it's just a blunder mistake, it is very difficult to find. I was actually puzzled myself. Excellent job! 

> Ideally it should at least pull the colors out from the Output window and/or Text Editor user preferences and leave the background colors up the the style sheet.

Yes, true. It can be achieved by exposing those colors as properties like [this](https://github.com/realthunder/FreeCAD/blob/a17b495c3e2be62679d2e526f067ea70c7a8556b/src/Gui/propertyeditor/PropertyEditor.h#L69). The difficult part though is how to deal with style changes where some style has this color specified, but some not. We'll need to somehow track the property setting per stylesheet and supply a sensible default when the property is not set. Do you want to give it a try?

closed 2021-08-23 by realthunder