realthunder/FreeCAD_assembly3 #790: Lofting two linear patterns of sketches only works when the sketches intersect

NothanUmber, 2021-08-23:
First - the possibility to use the linear pattern tool with sketches is a really great feature in the Readthunder branch!

My aim to use this is to create lofted solids by having two rows of patterned sketches opposing each other and to pair-wise loft them - with the number of pattern elements being a variable parameter.
By using different sketch sizes, positions and distances between the pattern instances, this can be used to e.g. create quite sophisticated channels inside a solid with a parameterizable number of such channels that a simple pattern of linear (drafted) extrudes cannot deliver.

FreeCad on the link branch is almost there with the functionality to create patterns of sketches. It even allows me to loft the two patterns. But apparently this only works if the sketches on at least one side intersect. If not I get the message "Loft: Failed to create shell" when trying to create a loft over patterns of non-overlapping sketches or "Recompute failed! Please check report view" if I change the pattern instance distance of initially overlapping sketches after the loft was already created.

Interesting thing is - the geometry that is generated in case of the overlapping sketches seems already to be exactly what would be expected to be generated in the case of the non-overlapping sketches. So it's perhaps just a sanity check that doesn't consider this corner case?

The images show:
[image]
* An example how such a with parameterizable channel-holes could look like (made in Onshape - but it is not the clean solution there either - as Onshape doesn't allow to loft the entire patterns either, I lofted the individual pairs manually. When reducing num_pipes to a number smaller than the prepared lofts those are simply shown as red (because the connected sketches were not generated) and when increasing the number above the number of prepared lofts then some hole-sketches simply are not connected with a hole. So a hack, but it works when num_pipes stays in a predictable range)
* An example from FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210717 that shows that the loft-of-patterns-of-sketches works as long as at least one side of the sketches overlap. It also shows a whole bunch of warnings related to TopoShape. Not sure whether these warnings mean that the topology was fixed or whether the fixing mechanism ran into issues. In the latter case this might be an interesting testcase for topo topics also.
[image]
[LoftedPatternsFreeCADExampleFile.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7033279/LoftedPatternsFreeCADExampleFile.zip)

realthunder, 2021-09-03:
Even when touched, the `Loft` does not produce a valid shape. I have added a feature to always show solid with double side lighting, so you can easily see an invalid. I have also added a new option `SplitProfile` to make this kind of intersected pipes work. It also works with non intersected case.

https://user-images.githubusercontent.com/1207888/132009551-8386f9e1-41bf-4cd1-a58b-945a39310111.mp4

closed 2021-09-03 by realthunder

NothanUmber, 2021-09-04:
Wow, this feature is super helpful! Thanks a lot!