realthunder/FreeCAD_assembly3 #792: Pad up to face (curved) does not seem to work between multiple bodies

klasrocket, 2021-08-24:
I seem to have a problem while trying to pad a sketch up to a curved surface that's belonging to a different body, that seems a bit weird.
[image]

[image]

I also tried with a flat surface instead and got another odd result, worked correctly when I added a -1mm offset to face:

[image]

but when I didn't use any offset I ended up with an offset in the model:

[image]

Hope these hints can be helpful, or maybe give me a hint if I missed something obvious.

[padtoface.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7038976/padtoface.zip)

I get the following error which does not tell me too much really:
14:17:15  <App> GeoFeatureGroupExtension.cpp(577): padtofacebodies#Pad001.UpToFace(links to padtofacebodies#Pad) is out of scope:padtofacebodies#Body vs. padtofacebodies#Body001

I also tried the pad up to curved face with shapebinders but does not get it to work and get similar errors.

OS: Ubuntu Impish Indri (development branch) (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

wohltat, 2021-08-24:
Your sketch plane (Sketch 002) is the wrong way around. The pad goes to infinity and returns on the other side :smile: 

Either rotate the line in Sketch001 that gives the direction or change "Map reversed" in Sketch002 (profile sketch for the pad)

klasrocket, 2021-08-24:
Thanks, that worked fine! Maybe not obvious but sort of makes sense.
I guess this could be communicated in a clearer way but not sure how. Guess I'll close this for now anyhow. Thanks for the help!

wohltat, 2021-08-24:
> I guess this could be communicated in a clearer way but not sure how.

That's true. I also just found this by accident.
Maybe it's still a bug and it shouldn't even be possible to pad / extrude in the opposite direction. From a real life point of view this would be reasonable.

klasrocket, 2021-08-24:
I agree, it would make more sense that it does not draw anything if the face is on the other side. I just noticed that the reverse option in the pad tools is inactivated when choosing up to face, but will still take effect if set to reverse when in dimensional pad, then change to up to face mode. This actually also makes my example work without flipping the sketch.

wohltat, 2021-08-24:
This indicates that is was meant to be used just in one direction by the author. Only all the checks haven't been implemented yet.

klasrocket, 2021-09-01:
Did some more testing and things actually behave as expected within a body but not when trying to pad in one body but using a reference surface in another. I guess this is not whats intended?

[image]

closed 2021-09-03 by realthunder