realthunder/FreeCAD_assembly3 #793: sub-shapebinder breaks thick solid tool

wohltat, 2021-08-24:
I wanted to use the workflow from a @oficinerobotica video (https://www.youtube.com/watch?v=N7b2LBgREmE). 
There the object is split apart by a surface and the splits are referenced via sub-shape binder. So both parts have the same origin but can be processed separately.
Now i want to create a shell with the thick solid tool. This works for the split but not for the reference of the split. The split is needed because the thickness tool needs a face of the inside, i.e. 

[image]
In the video i use a Loft but its the same with a simple cube.

I also tried shapebinder and Links but nothing works. I don't know how to reference the base object so that the thickness tool can be used. Also referencing before the split doesn't work either. 

To be able to use references for the thickness tool / splits would be very usefull for more organic shapes where the thickness tool would be easier than to model the shell manually.

realthunder, 2021-09-02:
The `SubShapeBinder` is intended to be used just like a `Datum` object. Although it has shape, it is not considered as part of the body shape.  And yes, using a `Boolean` feature can make it part of the body shape. You can also use various `Pattern` features on binder directly.

closed 2021-09-02 by realthunder

wohltat, 2021-09-02:
@dreetie Thanky, its works great. It seems to work also with more complex geometry with round / smooth surfaces. It is still limited to the pickiness / sensitivity of the thickness tool though. 
For some reason thickness outwards is more robust then inwards. I don't know why but it often fails on the object / on the split which has a seam. 

[image]

First i thought, its quite un-intuitive to fuse a an object with itself to achieve this. But with  @realthunder s explaination it makes more sense now. The reference is not part of the body shape and you are fusing it to the empty body shape.
Although still works and has an understandable reason, a new user would not come up with this solution by hirm self.

What about the thickness tool automatically fusing the shapebinder since it is clear from the selected faces which object is meant  to be modified which is not the tip of the body?