realthunder/FreeCAD_assembly3 #797: Part can't split solid with plane

parnasist3, 2021-08-28:
Can't split a solid with a plane, see picture and model.
I've tried "Shape splitting tools" and ordinary Booleans, but without success.
Any suggestions? (without resorting to dead copies, everything must be connected/linked as it is in Magic_hole_maker_Part_LinkDaily_2021.717.FCStd)

Disclaimer, this is just a general test for upcoming real and huge models.

[image]
[Magic_hole_maker_Part_LinkDaily_2021.717.FCStd.txt](https://github.com/realthunder/FreeCAD_assembly3/files/7070702/Magic_hole_maker_Part_LinkDaily_2021.717.FCStd.txt)

OS: openSUSE Leap 15.2 (KDE//usr/share/xsessions/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2

realthunder, 2021-08-29:
It is because the tool (Part::Plane) you are using is a finite plane. You need to make it big enough to slice through the object in order to make it work. Alternatively, you can use the `Datum plane`, which provides an infinite plane.

[image]

parnasist3, 2021-08-29:
> ... (Part::Plane) ... is a finite plane.
 
Oh com'on! You can't seriously have a plane that's not infinite in your program. That is sacrilegious and in breach of all rules!

Replace that finite plane with the normal/standard infinite one.
Or rename it to what it is, "Finite flat surface". Then add the proper infinite plane and denote it just that, "Plane".

Or even better, tidy up the whole mess of points, lines, curves and planes -> one pop-up window per what the user want to create.
One for point creation.
One for line creation.
One for curve creation.
One for plane creation.

luzpaz, 2022-04-17:
@realthunder verdict on this ticket?