realthunder/FreeCAD_assembly3 #798: unable to export model ("body has null shape")

NothanUmber, 2021-08-29:
When trying to export the following model as STEP or STL I get the warning message "body has null shape". In the case of the STEP export nothing happens (no file is created) and for STL export I get a file with this content:
MESH-MESH-MESH-MESH-MESH-MESH-MESH-MESH-MESH-MESH-MESH-MESH-MESH-MESH-MESH-MESH

Don't see what is wrong with my FreeCAD file on first sight.
[PolyMoeRT.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7072674/PolyMoeRT.zip)

realthunder, 2021-09-02:
The problem is that you did something that somehow unset the body tip (see the screencast on the visual representation of the tip, the green dot inside the tag icon). The body tip marks which shape to export for that body. This is certainly a bug, but hard to pinpoint exactly what you've done to cause the loss of tip. I have add a patch to check for tip when the body is recomputed.

BTW, a feature probably not many knows, you can export more that one shape by right click the feature and choose `Toggle export`. This is added because you can have multiple solids inside a body.

https://user-images.githubusercontent.com/1207888/131769125-45c60dbd-8fd4-491e-97d8-b77883cd43d0.mp4

closed 2021-09-02 by realthunder

NothanUmber, 2021-09-02:
Thanks for the "set body tip" feature - good to have an option to repair a project when much work went into it.
Would of course be better if this cannot happen in the first place. Will open a new issue when I should find steps that reproducably bring an object into that state.