realthunder/FreeCAD_assembly3 #799: 'Revolve a selected sketch' tool sometime make mistake in part design workbench.

LeoBinsc, 2021-08-29:
'Revolve a selected sketch' tool sometime make mistake in part design workbench.
i have two case to show the mistake.
one case.
i create first sketch and draw a line Parallel to y axis.
next, i create second sketch and draw a rectangle.
then i use 'Revolve a selected sketch' tool to create a ring.
but sometime it work,sometime it makes mistake .

another case .
i pad a new sketch which draw a rectangle.
i select the top face and use 'Revolve a selected sketch' tool, but it has wrong result .
it sames nothing has changed.

i use the latest version.
sometime the tool is work well, but not all the time.
a man told me the master version work well.
maybe we should check the difference of this tool between this two version.
and some picture and gif  which shows detail are in this website.
https://forum.freecadweb.org/viewtopic.php?f=8&t=61714

luzpaz, 2021-08-30:
@LeoBinsc please add you full [About](https://wiki.freecadweb.org/About) info to the OP

LeoBinsc, 2021-08-30:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git)
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Chinese/China (zh_CN)

i find more detail about groove.
groove have mistake when deal with same place twice.
https://forum.freecadweb.org/viewtopic.php?f=20&t=61714&start=20

realthunder, 2021-08-31:
Hmm... this is quite a curious case. I find that `Revolve` and `Groove` misbehave if I just use the sketch in your posted file (attached below). But if I manually recompute your sketch once, then both `Revolve` and `Groove` works as expected. There must be some very obscure bug hidden somewhere causing the sketch shape in some `invalid` state, so to speak. Need further investigation. Can you think of anything special you've done with the sketch (the one with the rectangle)?

[part design019.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7084354/part.design019.zip)

LeoBinsc, 2021-08-31:
part design is also wrong result.
the bottom of ring should be air core.
but the center circle is  surface.
it should be a Hollow ring.
if your program have right result , maybe our environment of computer are different.
just said this by part design 19.
my other files shows the program of revolve and groove have problems.

LeoBinsc, 2021-08-31:
rectangle in xy sketch which raise above 20mm in z order , is also has wrong result after revolve by x axis  turn 90 degree.
i will talk about it .
the  Correct cross section is next picture.
[image]
but do this ,you will have wrong result.
[image]
[image]
[image]
turn order with x axis has more issues than y axis.

LeoBinsc, 2021-09-01:
when z >0 or z < 0, revolve the sketch ( with x axis) which rectangle Intersects the X axis in the  top view, the program allow, but sometime Not handled well. because it is more difficult ,maybe have Overlapping area to Calculation and display.
when z=0,revolve the sketch which rectangle Intersects the X axis in the  top view, the program forbid this situation.
so delay solve it, and wait new OCC version is a good idea.

realthunder, 2021-09-01:
There is something weirdly wrong with your FreeCAD, settings maybe. The saved sketch shape seems messed up. It can't be used to do any operation, even for a simple `Pad`. Please post your settings in `C:/users/<you_user_name>/AppData/Roaming/FreeCAD/link.user.cfg`.

LeoBinsc, 2021-09-01:
[link.user.cfg.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7087880/link.user.cfg.zip)
i use master and linkstage3  in my computer.
so  in the folder,there has all configuration file.
i just have changed mod folder.

LeoBinsc, 2021-09-01:
[image]
this picture show the same as rectangle in xy sketch which raise above 20mm in z order.
turn around this rectangle have Overlapping area to Calculation and display.

LeoBinsc, 2021-09-01:
i think the key is ,if  revolve the sketch  with x axis ,the pattern should be On one side of the x-axis, which means y>0 or y<0 all of the pattern. if the pattern contain y > 0 and y<0, there will have have Overlapping area.
if it is difficult to program,  delay handle it and look other software is the best way.
[image]
[image]
[image]

LeoBinsc, 2021-09-01:
revolve the xy sketch(z=0) with x axis, one pattern can not  connection first with third or fourth quadrant ,also can not  connection second with third or fourth quadrant.
revolve the xy sketch(z dot equal to 0) with x axis should have same forbid.

realthunder, 2021-09-01:
I still can't reproduce your problem with your setting. I also tried with the exact same Window release image. Maybe you didn't get what I mean by recomputing the sketch. Please check out the screencast below. Without manually recompute the sketch first, I can't use it to do anything. Even a simple `Pad` will fail. But are you telling me you can use it fine, I mean apart from some Revolve/Groove anomaly?

https://user-images.githubusercontent.com/1207888/131646445-f73c4cc1-b6ef-4585-aeea-b7770b306dc9.mp4

LeoBinsc, 2021-09-01:
you just raise the xy sketch up 20mm , it is necessary to select x axis to revolve.
because when z = 0, the program announce revolve x axis intersects the sketch. (of course use this pattern)
and do step my picture shows.
[image]
[image]

LeoBinsc, 2021-09-01:
the program can not solve the revolve which create Overlapping area , maybe it is useless.
so i suggest forbid this situation.

LeoBinsc, 2021-09-01:
it is work well except the situation of raise  sketch up and revolve the sketch (which cross over x axis ) with x axis.
add a exception to solve this situation.
because this situation is useless.

LeoBinsc, 2021-09-01:
and even this situation can be split the pattern with x axis, and move  upper pattern a big distance toward x direction.
then you can find there has Overlapping area In the X direction . just delete the distance of upper pattern in the x direction.
the Overlapping area Synthesize one，then it is the correct result。
but it is  useless.

LeoBinsc, 2021-09-01:
i put a gif to show this mistake.
[image]

LeoBinsc, 2021-09-06:
suggest
[image]