realthunder/FreeCAD_assembly3 #802: tree view indentation setting

bambukouk, 2021-09-10:
as per:
https://forum.freecadweb.org/viewforum.php?f=3

how do i set value for tree view items indentation, please?

bambukouk, 2021-09-12:
Took me two days, but eventually remembered... 👍 
Tools>edit parameters
Would have (should have) asked this question on FreeCAD forum, but the they do not want to talk to me unless I use "master" branch 👎

closed 2021-09-12 by bambukouk

realthunder, 2021-09-13:
Sorry for not responding earlier. Do you use discord? Someone has set up a new [channel](https://discord.gg/NpMefpXWFT) for my branch. You can ask there next time.

bambukouk, 2021-09-13:
Don't apologise 😉
It was my stupidity and bad memory...
You have enough work dealing with real FreeCAD issues
Thank you for all your work!