realthunder/FreeCAD_assembly3 #803: cyclic dependency for sub-shape binder

bambukouk, 2021-09-10:
I am having issues creating sub-shapebinder for bodies inside assemblies.
Yet there are no problem creating shape binders using the same bodies?
Also there are no problems creating sub-shapebinders for the same bodies if outside assembly
Do not understand this? - can someone, please explain/iluminate?
file in question:
https://drive.google.com/file/d/1RDNJsc70C-V0dp6lG6y-Q5ZSE0oHPkgf/view?usp=sharing
and what I am trying to do:
[image]

i.e. sub-shape link inside active "exhaust body" using "cylinder body"

I have asked this question on FreeCAD forum, but have been told:
"If this depends on Assembly3, then you should ask this in the Assembly forum, or I can move the topic there. If not, please reproduce it in master. Please note that realthunder is distributing a different program; it is not FreeCAD."

bambukouk, 2021-09-10:
Later edit:
Found a solution!
In order to be able to create sub-shape binder in "exhaust_body" I temporarily removed "Ass_exhaust_tee" from the top level assembly - no longer cyclic dependency !!!
Having created sub-shape binder I have put "Ass_exhaust_tee" back as a part of top level assembly and there it is, with sub-shape binder added:
[image]

I remain puzzled :-)

realthunder, 2021-09-22:
There is indeed a related bug that's has been fixed recently. Can you please try using the latest release?

bambukouk, 2021-09-22:
Thank you!
Traveling back home tomorrow.
Will install latest release and test asap.

bambukouk, 2021-09-24:
Yes!
Tested and all is working as it should :+1: 
Problem solved.
Thank you very much indeed, yet again!

closed 2021-09-24 by bambukouk